schodet/nxt-python #69: Keep getting BrickNotFound error

TheDasMan, 2015-11-18:
I installed the package according to all the instructions. I installed PyUSB and Libusb. But whenever I try to use anything, it throws the BrickNotFound error. I'm running Windows 7 64 Bit, NXT with the newest firmware. I'm connecting by USB. And I'm on Python 3.5. I don't know what else I should include, but that's as much as I can think of.

Eelviny, 2015-11-18:
This is designed for use on Python 2.7. Please read the installation guide [here](https://github.com/Eelviny/nxt-python/wiki/Installation).
If you're trying to pull the code directly from the github repo, it's currently in an unusable state.

TheDasMan, 2015-11-19:
So do I download the "Download Zip" on the main page? Sorry, I'm really new to using github

Eelviny, 2015-11-21:
You need to pick up the latest release on the [releases](https://github.com/Eelviny/nxt-python/releases) page.
Also, I'm still figuring out everything myself (I recently took over this project) and I'm not sure USB connection can do more than send and receive files. Bluetooth should work flawlessly.

Eelviny, 2016-01-28:
Closing due to inactivity

closed 2016-01-28 by Eelviny