schodet/nxt-python #70: Import nxt.locator    --  USB Error

lkvapil, 2016-01-11:
Hello all,

I need help with my Raspberry Pi 2 and NXT (firmware 1.31).
I am trying to connect NXT to raspberry and it shows only errors with nxt.locator. 
Same error shows, when I try it same with Pc - Windows 7. I did same steps as at official github...
I paste here some link to images that I upload... I write before all commands sudo and I installed os across NOOBS.
I tryed too disconnect all usbs from pi usb hub and only connect nxt. No change. I spend lot of my free time and nothing..

http://lukas263.imgur.com/

Thanks for all answers..

All the best
Lukas

Eelviny, 2016-01-28:
Sorry for taking my time over this... I try to dedicate time to this project but life gets in the way.
I've yet to sort usb issues, as this is to do with a major update to the library libusb that I haven't updated for. 
If you absolutely must get it to work, try looking into downgrading from libusb version 1.0 to version 0.1, but I'm not sure how to go about this on a raspberry pi. 
I'm no longer supporting version 2, but thanks for the errors! Should be useful for updating v3.

_For future reference, instead of screenshotting errors, if you could copy and paste them inside a code block (Read up on the [Github Markdown](https://guides.github.com/features/mastering-markdown/) then it's much easier for me to see them._

AlanJAS, 2016-03-23:
This error http://i.imgur.com/m4327EP.jpg
Is that you try use python 2.7 with the latest code. The code was migrated to Python 3. You must use:
python3 mary.py (with sudo if not have the udev rules).
For PyUSB I recommend use the latest tagged version 1.0.0rc1 from https://github.com/walac/pyusb
Clone it and put the folder "usb" in the folder "nxt". And remove the old python-usb package that you install.

AlanJAS, 2016-03-23:
Other way: use this version https://github.com/AlanJAS/nxt  of nxt, with latest PyUSB. This code was tested with Python 2.7 and works.

Eelviny, 2016-03-23:
Whoops, didn't mean to close this. @AlanJAS, I've merged your commit, but I have no time for testing. If you confirm that it is working as intended, I will mark this as fixed.

lkvapil, 2016-03-24:
Thanks, I will try it and write back...

lkvapil, 2016-04-01:
So..

The codes works fine, but only in the moment, when the NXT starts up :D ???
I dont know, always writes error.. -- I shut down NXT, then I press orange button to start up it and 
immediately I press enter to run the program (code) and it works fine. But... 15 seconds after start up it doesnt work.. :(((

lkvapil, 2016-04-01:
---I selected option from AlanJAS 

> Other way: use this version https://github.com/AlanJAS/nxt of nxt, with latest PyUSB. This code was tested with Python 2.7 and works.

lkvapil, 2016-04-15:
For all here.. I have tested the os from list here installed with berryboot.. `https://sourceforge.net/projects/berryboot/files/os_images/` and successful I run NXT-PYTHON with version Raspbian Jessie 2015.9 `https://sourceforge.net/projects/berryboot/files/os_images/Debian_Jessie_Raspbian_2015.09.img192/download` Thanks..

feinstaub, 2016-04-22:
Why is the label "component:bluetooth" added here?

Eelviny, 2016-04-22:
Just... pretend it was usb all along :P

feinstaub, 2016-04-22:
> Whoops, didn't mean to close this. @AlanJAS, I've merged your commit, but I have no time for 
> testing. If you confirm that it is working as intended, I will mark this as fixed.

See here https://github.com/Eelviny/nxt-python/issues/72#issuecomment-213442591 and here https://github.com/Eelviny/nxt-python/issues/72#issuecomment-213567186: maybe https://github.com/Eelviny/nxt-python/commit/0daed0696d38c14ffb06582cb7f4e60b71a67515 should be reverted...

feinstaub, 2016-04-22:
For reference: I use python3-usb 1.0.0-5.2 (openSUSE 42.1 package). Python 3.4.1. NXT (Protocol version 1.124, Firmware version 1.31)

Eelviny, 2016-04-22:
Ticket #83 will now hold all discussion pertaining to the updates and fixes to the USB implementation.

dlech, 2018-04-28:
I think this issue can be closed. The original error looks like an error because of using python2 after the python3 conversion.

commit d94bccc0 (2016-03-23): Merge pull request #76 from AlanJAS/master

adapt usbsock to latest PyUSB 1.0.0rc1 Fixes #70
closed 2021-11-05 by schodet