schodet/nxt-python #71: Test script not Py3 compatible

Pixdigit, 2016-01-24:
The nxt_test in scripts is not python3 compatible, while some required libraries are py3 exclusive. Please correct all print statements, so they have "((" and "))".

Eelviny, 2016-01-28:
Whoops, didn't see that one. I'll give that a quick fix up.
Right now the bigger problem is v3 doesn't work at all, but hey, since I'm here already ;)

closed 2016-01-28 by Eelviny

Pixdigit, 2016-01-29:
Turns out ipsock.py, brick.py, bluesock.py, usbsock.py and locator.py are not Py2 compatible … ?
--> http://pastebin.com/iZt2miDh

Eelviny, 2016-01-29:
Everything in the master branch should by Py3 compatible. Everything in the Python 2 branch, and the latest releases, should be Py2 compatible.
Which are you using? You should be using the latest release.

Pixdigit, 2016-01-29:
It was pulled directly from the https://github.com/Eelviny/nxt-python repo.

Eelviny, 2016-01-29:
That'll be why.
1. The latest commits are designed exclusively for Python 3
2. The repository is currently unstable to the point of unusable, and you should use the versions on the releases page.

Pixdigit, 2016-01-29:
Yes, that should definitely solve the problem. But to avoid further irritation the Master branch could be marked as unstable and the Py2 branch as preferred source since it is best practice to have the preferred version in Master and make branches for in development versions. (Just pointing out. I don't do so really either)

commit 0ee0a9df (2016-01-28): Add brackets around print statements. Fixes #71