schodet/nxt-python #72: Problem connecting (TypeError: 'Telegram' object is not iterable)

richrd, 2016-01-26:
```
Connecting...
Host: 00:16:53:02:XX:XX Name: None Strict: True
USB: True BT: True Fantom: False FUSB: False FBT: False
USB module unavailable, not searching there
Traceback (most recent call last):
  File "/usr/local/lib/python3.2/dist-packages/nxt/locator.py", line 131, in find_one_brick
    info = b.get_device_info()
  File "/usr/local/lib/python3.2/dist-packages/nxt/brick.py", line 27, in poll
    self.sock.send(bytes(ogram))
TypeError: 'Telegram' object is not iterable
Failed to connect to possible brick
```
I converted my working script for Python2 to Python3. Now I'm unable to connect to to the brick using the same hardware as before. The host is also correct. Am I doing something wrong, or is this a bug?

Ps thanks for the awesome lib :)

richrd, 2016-01-27:
I was in a hurry when submitting this so I didn't notice that v3 isn't usable yet. Anyway I tried to fix this myself, but I don't know enough about NXT bluetooth communication to understand how the messages should be formatted. 

I'd be willing to try to help porting `bluesock.py` if I someone would point me in the right direction.

Eelviny, 2016-01-28:
Hi richrd,
If you've got some Python experience, I'd be glad for some extra hands on board! The problem is these days, the NXT is quite old and all the online resources detailing the communication protocol are slowly fading away...
The problem goes right back to the [telegram.py](https://github.com/Eelviny/nxt-python/blob/master/nxt/telegram.py) module, which I'm struggling to wrap my head around myself, as this is the section which outputs the raw commands the NXT listens for. I've already had a poke around the `blusock.py` section, and spotted nothing obvious there...

richrd, 2016-01-28:
I'm pretty familiar with Python, but don't have experience with `pack` and `unpack` used in telegram.py.
On the other hand I found some specs for NXT communication, but you've probably seen them already. https://code.google.com/archive/p/smart-robot/downloads . I re-uploaded them here just incase: http://bittemple.org/misc/nxt/ .

I might try to look into telegram.py later when I have time.

guihi, 2016-02-09:
Hi together, I could fix this. The problem was the mixture of bytes and String.
It needed modifications in three places:
Here - in breve - my modifications:
usbsock.py
    def recv(self):
<77..>
        # NOTE: bulkRead returns a tuple of ints ... make it sane
        **return bytearray(data)**
        #return ''.join(chr(d & 0xFF) for d in data)

telegram.py
<58..>
    def __str__(self):
        return self.pkt.getvalue()
	# doesn't work this way because value is bytes!!

    **def bytes(self):
        return self.pkt.getvalue()**

brick.py
<26..>
        with self.lock:
            self.sock.send(**ogram.bytes()**)
            if ogram.reply:
                igram = Telegram(opcode=opcode, pkt=self.sock.recv(

raph38130, 2016-02-16:
I have an issue with the third edit...
brick.py
<26..>
with self.lock:
    self.sock.send(ogram.bytes())
    if ogram.reply:
        igram = Telegram(opcode=opcode, pkt=self.sock.recv(

there is no bytes method in Telegram
(should be str() ?)

rgds

raph38130, 2016-02-16:
I got it !!! try this :

import bluetooth
sock = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
sock.connect(("00:16:53:02:B8:D5", 1))
print("connected")
#data=(LSBsize,MSBsize,type,CMD,data)     (if some data is a word = lsb, msb)
#data=bytearray((2,0,0x01,0x9B))    #get NXT brick name
#data=bytearray((2,0,0x01,0x88))    #get fw version
data=bytearray((6,0,0x00,0x03,128,1,0,4)) # play sound

sock.send(bytes(data))
print("sent")
buf=sock.recv(16)
print("received"+str(buf))
print("end")

guihi, 2016-02-16:
Sorry, but my comment was formatted ugly. So the important new part for telegram.py did look different (should have been bold). 
You have to add a function in telegram.py (see above):
`def bytes(self):`
`    return self.pkt.getvalue()`

Eelviny, 2016-02-19:
I've taken your changes and put them into a separate branch - if you're happy with the changes, I'll merge to master.
Got no time to test at this point in time, sorry

Eelviny, 2016-02-19:
Please continue any conversation about writing code at #75

guihi, 2016-02-19:
Thank you very much for your work!
I hope to learn better the formatting of comments.
Next week I will be able to test more of the code and give you a return. I need nxt-python for some exercises in the college where I'm teaching. About the code I will follow your advice and write at #75

feinstaub, 2016-02-26:
With your "bytes" branch I managed to connect to brick. However the ** message_test.py** from the example repository fails with:

```
Traceback (most recent call last):
  File "message_test.py", line 11, in <module>
    b.message_write(box, 'message test %d' % box)
  File "/home/gregor/.local/lib/python3.4/site-packages/nxt/brick.py", line 25, in poll
    ogram = poll_func(opcode, *args, **kwargs)
  File "/home/gregor/.local/lib/python3.4/site-packages/nxt/direct.py", line 113, in message_write
    tgram.add_string(len(message), message)
  File "/home/gregor/.local/lib/python3.4/site-packages/nxt/telegram.py", line 68, in add_string
    self.pkt.write(pack('%ds' % n_bytes, v))
struct.error: argument for 's' must be a bytes object
```

Eelviny, 2016-02-26:
@feinstaub, just made a new commit to the bytes branch, have a go now

feinstaub, 2016-02-26:
Thanks for quick response. I will be able to test it in one week time. I'll report back then.

Eelviny, 2016-03-24:
@feinstaub, were you able to give it a test? Mine are coming out fine, but good to get confirmation from multiple sources.

feinstaub, 2016-03-24:
No, Sadly, I didn't get in touch with the robots yet. I'll give feedback. Is the change in master now or do I still have to use the "bytes" branch?

Eelviny, 2016-03-24:
Yeah, all moved over to `master`. `bytes` branch has been deleted.

feinstaub, 2016-03-24:
Cool.

richrd, 2016-03-24:
Hi again! Great to see all the attention this got, and sorry I've haven't been able to test this yet. Thanks very much for work, I'll test this before monday.

Keep it up! :)

feinstaub, 2016-04-22:
Hi. Thanks for asking. I am trying today again. Currently, I have a strange problem: the nxt.locator.find_one_brick does not find my brick anymore. I tried to add more debugging to the locator.py file by adding a print statement

```
def find_bricks(host=None, name=None, silent=False, method=Method()):
    """Used by find_one_brick to look for bricks ***ADVANCED USERS ONLY***"""

    print("aaa")
    sys.stdout.flush()
```

At the end of the file I added:

```
def my_test():
    print("hallo")

print("start")

my_test()
find_bricks()
```

Now if I run the file with python the output is:
```
start
hallo
```

and nothing more.

feinstaub, 2016-04-22:
There is an error in usbsock.py

```
b = nxt.locator.find_one_brick(None, None, False, None, True)
Warning: Config file (should be at /home/gregor/.nxt-python) was not read. Use nxt.locator.make_config() to create a config file.
Host: None Name: None Strict: True
USB: True BT: True Fantom: False FUSB: False FBT: False
aaa
bbb
Connecting via USB... 
Connected. 
Traceback (most recent call last):
  File "/home/g/.local/lib/python3.4/site-packages/nxt/locator.py", line 140, in find_one_brick
    info = b.get_device_info()
  File "/home/g/.local/lib/python3.4/site-packages/nxt/brick.py", line 27, in poll
    self.sock.send(ogram.bytes())
  File "/home/g/.local/lib/python3.4/site-packages/nxt/usbsock.py", line 69, in send
    self._debug('Send:', end=' ')
TypeError: _debug() got an unexpected keyword argument 'end'
Failed to connect to possible brick
```

It is caused because this method:

```
def _debug(self, message, err=''):
```

is called like this:

```
self._debug('Recv:', end=' ')
```

There is no end parameter.

feinstaub, 2016-04-22:
After fixing the above error (there are two places where I removed the end='   ') I got this error:

```
Traceback (most recent call last):
  File "/home/g/.local/lib/python3.4/site-packages/nxt/locator.py", line 140, in find_one_brick
    info = b.get_device_info()
  File "/home/g/.local/lib/python3.4/site-packages/nxt/brick.py", line 27, in poll
    self.sock.send(ogram.bytes())
  File "/home/g/.local/lib/python3.4/site-packages/nxt/usbsock.py", line 70, in send
    self._debug(':'.join('%02x' % ord(c) for c in data))
  File "/home/g/.local/lib/python3.4/site-packages/nxt/usbsock.py", line 70, in <genexpr>
    self._debug(':'.join('%02x' % ord(c) for c in data))
TypeError: ord() expected string of length 1, but int found
Failed to connect to possible brick
```

Something seems to be wrong with parsing the received data.

feinstaub, 2016-04-22:
The error is caused by this commit 0daed0696d38c14ffb06582cb7f4e60b71a67515.

Reverting to 9c80907a6c67e8262ce8aa3732591edf6b7493e5 fixes the problem and 

~/.local/bin> ./nxt_test 

yields

```
NXT brick name: b'NXT\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
Host address: 00:16:53:05:D2:A2
Bluetooth signal strength: 0
Free user flash: 16476
Protocol version 1.124
Firmware version 1.31
Battery level 7477 mV
0]gregor@linux-85rg:~/.local/bin> ./nxt_test 
NXT brick name: b'NXT\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
Host address: 00:16:53:05:D2:A2
Bluetooth signal strength: 0
Free user flash: 16476
Protocol version 1.124
Firmware version 1.31
Battery level 7477 mV
```

Eelviny, 2016-04-22:
The old code causes PyUSB to run on a compatibility layer which is not recommended... but of course the new implementation is not quite there yet. @feinstaub, what I'll do is revert the changes on `master` branch, and create a new branch for it called `pyusb`. 
The PyUSB implementation needs high priority work as this compatibility layer isn't going to be around forever.

feinstaub, 2016-04-22:
Sounds good!

Eelviny, 2016-04-22:
PyUSB implementation reverted in 6a7992301cf1ba7de5fb098df9901c9d3c643794.

Eelviny, 2016-04-22:
I've created a new ticket #83 to hold all details pertaining to the update of PyUSB, since it's causing many similar issues that need to be consolidated.

dlech, 2018-04-28:
I'm not seeing this issue either, so this issue can probably be closed.

GolddolphinSKB-Fexcraftian, 2018-04-30:
Issue fixed, I guess.

closed 2018-04-30 by GolddolphinSKB-Fexcraftian