schodet/nxt-python #74: Cannot connect to NXT via USB on Linux

Pixdigit, 2016-01-29:
http://pastebin.com/59mKPF4M

Since these are classes referring to other classes I have no Idea what is really going on. (I got stuck at finding out, what the brick's socket is, since its only initialized with "self" argument)

feinstaub, 2016-02-26:
I got a similar error on openSUSE 42.1 and python3. Your output suggests that you use python2.7.

The first error on your output is this:
TypeError: write() takes at most 4 arguments (5 given)

My output says

TypeError: 'Telegram' object is not iterable

Full output:
```
Warning: Config file (should be at /home/u1/.nxt-python) was not read. Use nxt.locator.make_config() to create a config file.
Host: None Name: None Strict: True
USB: True BT: True Fantom: False FUSB: False FBT: False
Traceback (most recent call last):
  File "/home/u1/.local/lib/python3.4/site-packages/nxt/locator.py", line 131, in find_one_brick
    info = b.get_device_info()
  File "/home/u1/.local/lib/python3.4/site-packages/nxt/brick.py", line 27, in poll
    self.sock.send(bytes(ogram))
TypeError: 'Telegram' object is not iterable
Failed to connect to possible brick
Bluetooth module unavailable, not searching there
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/u1/.local/lib/python3.4/site-packages/nxt/locator.py", line 149, in find_one_brick
    raise BrickNotFoundError
nxt.locator.BrickNotFoundError

```

feinstaub, 2016-02-26:
I just saw that the type of error I got is already discussed here: https://github.com/Eelviny/nxt-python/issues/72

Eelviny, 2016-02-26:
I believe this is a duplicate of #72, closing

closed 2016-02-26 by Eelviny