schodet/nxt-python #77: works exactly once, after that it throws BrickNotFoundError

oliverkurth, 2016-04-04:
Hi,
I installed nxt-python, version 2.2.2-2 in a Ubuntu 15.10 vm, and connected a nxt with USB. I tried the spin.py program from the examples and it works, but only once. When I try it again, I get the exception BrickNotFoundError:

vmware@ubuntu:~/python-nxt$ ./spin.py 
vmware@ubuntu:~/python-nxt$ ./spin.py 
Fantom module unavailable, not searching there
Traceback (most recent call last):
  File "./spin.py", line 12, in <module>
    b = nxt.locator.find_one_brick()
  File "/usr/lib/python2.7/dist-packages/nxt/locator.py", line 137, in find_one_brick
    raise BrickNotFoundError
nxt.locator.BrickNotFoundError
vmware@ubuntu:~/python-nxt$ 

When I unplug the nxt and plug it in again it works again, but only once.

I configured the name and host id in ~/.nxt-python, but it makes no difference.

It's an old NXT, in case that matters. FW version is 1.03.

lkvapil, 2016-04-18:
I have the same error

oliverkurth, 2016-04-19:
It works fine on a Raspberry Pi, but not in a VMware VM. @lkvapil what is your environment?

Eelviny, 2016-04-19:
Strange, very strange... @oliverkurth, If you've got a USB bluetooth dongle around, are you able to test a bluetooth connection with VMWare?

lkvapil, 2016-05-09:
What did you mean by that ? - Raspberry Pi 2, Raspbian 2015.09

Eelviny, 2016-05-11:
@lkvapil, what @oliverkurth means by "environment", is your operating system and software versions.

e.g. I'm running Ubuntu 14.04 with Python 3.5.1 and nxt-python 2.2.2-2

vishen89, 2017-05-10:
> I configured the name and host id in ~/.nxt-python, but it makes no difference.
Hey I am not able to find the file "~/.nxt-python" to edit. How do I find it?
I did generate it, but how to I open and edit the file?

mattschueler, 2018-05-20:
I have gotten this issue sometimes when running my code, I think it is caused by not "closing" the connection properly, because it happens only when running a second time. Only way I have been able to fix it is by restarting my kernel or unplugging and plugging the NXT back into my PC. I would suggest running with debug=True and seeing what the stack trace says, it's usually pretty informative.

schodet, 2021-11-16:
Should be fixed by the `reset` PyUSB call now. Please reopen if you can still reproduce with master branch and python3.

closed 2021-11-16 by schodet