schodet/nxt-python #78: NXT-PYTHON in browser

lkvapil, 2016-04-13:
Hi,

my idea is to control Lego NXT by NXT PYTHON over the browser. I want to make it 
easiest, but without lines with commands type to write into the console and start the PYTHON script, next start other script .... I mean all into the Python with buttons.. (I dont know, if somebody all ready done it pls. tell :D) Is here somobedy able to spend free time with my idea ? Thanks

nikolajovickg, 2016-04-19:
Done that with node.js and python, but with console. You can implement what you need with nxt-python and socket.io (and serve static web page with python).

lkvapil, 2016-04-20:
Could you leave some url ? :D

schodet, 2021-11-16:
Out of scope, you can build a project using NXT-Python to do this though, using a backend in python and a web page communicating with the backend.

closed 2021-11-16 by schodet