schodet/nxt-python #83: Complete PyUSB v1.0 implementation

Eelviny, 2016-04-22:
The current PyUSB implementation is horribly outdated, and while it works, it must be updated.
I'm creating a new ticket to specifically detail progress on the `usbsock.py` module, as many tickets are arising that are based on this issue.

All changes to `usbsock.py` will now be committed to the `pyusb` branch.

petterreinholdtsen, 2017-08-01:
Hi.  I'm one of the Debian maintainers of nxt-python, and wonder if you have a timeline for when the pyusb API usage will be updated?  Would be nice if a fix for this was included in the new Debian release.

Eelviny, 2017-10-15:
Hi @petterreinholdtsen,

As of right now, this project is not being updated. While I am maintaining the documentation and issue tracker, I'm afraid I currently have no plans to continue development on this due to personal commitments.

closed 2021-11-10 by schodet