schodet/nxt-python #85: nxt_filer

feinstaub, 2016-04-22:
The nxt_filer script is missing a description of what it is supposed to do.

Is it still needed? (on openSUSE 42.1 there is only Python2 version of the required gtk package so I am not able to run it)

Eelviny, 2016-04-22:
The script opens a GUI that allows for file dragging and dropping to and from the NXT brick.
A very useful tool, but it needs a complete rewrite, as the `gtk` module is outdated and PyGObject is needed.
I'll delete it, as it's still included in the `python2` branch.

I'm going to keep this issue open in the hopes that someone will take it upon themselves to give it the rewrite it deserves.

Eelviny, 2016-04-22:
See commit f4b47489b5a499dd2a65f11cf51c5ff3a0b47f9c

tcwan, 2018-03-09:
@Eeviny,

I think you can close this issue, nxtfilemgr is the replacement.

feinstaub, 2018-04-21:
I closing this issue, see also https://github.com/Eelviny/nxt-python/pull/133

closed 2018-04-21 by feinstaub