schodet/nxt-python #88: nxt-python on Android ?

pmu5757, 2016-04-29:
Hello,
I would like to know if nxt-python works on an Android device.
It would be fine because of accessing to all the phone's sensors and functions  (GPS, Accelero, SMS, camera...)
Thank you.

Eelviny, 2016-04-29:
Hi,

Sounds like an interesting idea you've got there! However, I'm going to have to say no - not without writing whole new sections of the nxt-python.
Bluetooth for example uses the bluez library to connect. This library is not used on Android, so it will need a new method of communicating with the Bluetooth API. For USB, it's a lot more difficult on Android, and again you run into API problems - This uses libusb, whilst Android will have its own API for USB interaction.

It's certainly possible, don't get me wrong - the best solution would be to fork NXT-Python, and create new bluetooth or usb libraries to interface with the Android API.

pmu5757, 2016-05-01:
Hi,
Thank you for your clear answer.
I'll keep on looking for another simple way to control the NXT with an Android device :-)

closed 2018-02-27 by Eelviny