schodet/nxt-python #92: Objc: module object has no attribute 'setSignatureForSelector'

roechi, 2016-10-11:
Hi, I am new to this and trying to figure out the following problem:

Running nxt_test (or any other script for that matter) results in the following error. Apparently there is an issue with pyobjc. Im running Python 2.7.12 and lightblue 0.4 on OSX 10.12.

Error while running test:
  File "nxt_test", line 25, in <module>
    b = nxt.locator.find_one_brick(debug=debug)
  File "/usr/local/lib/python2.7/site-packages/nxt/locator.py", line 112, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "/usr/local/lib/python2.7/site-packages/nxt/locator.py", line 51, in find_bricks
    import bluesock
  File "/usr/local/lib/python2.7/site-packages/nxt/bluesock.py", line 18, in <module>
    import lightblueglue as bluetooth
  File "/usr/local/lib/python2.7/site-packages/nxt/lightblueglue.py", line 17, in <module>
    import lightblue
  File "/usr/local/lib/python2.7/site-packages/lightblue/__init__.py", line 160, in <module>
    from _lightblue import *
  File "/usr/local/lib/python2.7/site-packages/lightblue/_lightblue.py", line 27, in <module>
    import _IOBluetooth
  File "/usr/local/lib/python2.7/site-packages/lightblue/_IOBluetooth.py", line 61, in <module>
    objc.setSignatureForSelector("IOBluetoothSDPServiceRecord",
'module' object has no attribute 'setSignatureForSelector'

Thanks for any help.

tcwan, 2016-12-28:
There were some Bluetooth driver changes in MacOS since 10.8, which results in the error you see.
I think there's no solution for using lightblue on MacOS 10.12 unless there are newer patches available.
Take a look at my comments in https://github.com/Eelviny/nxt-python/issues/95

dlech, 2018-04-28:
Bluetooth on macOS should be fixed by #147. You will need to create a config file such as this:

```
[Brick]
name = NXT
host = 00:16:53:00:00:00
strict = 0
method = usb=True, bluetooth=False, device=True
```

This will tell nxt-python to use the Bluetooth serial port created by macOS automatically (this is what `device=True` is for). No python bluetooth packages required.

closed 2021-11-08 by roechi