schodet/nxt-python #93: SyntaxError: invalid syntax

guruhj, 2016-10-26:
Hi, I am using Python version 2.7.11 and python-nxt verstion 2.2.2. in Windows 7.
I have installed PyBluez, PyUSB and Pyfantom without any error message.
But, when I try to test for any code from example folder, I always get error messages below. 
Please help me how to handle this problem. Thanks in advance. 
 
=============== RESTART: C:\nxt-python-2.2.2\examples\spin.py ===============

Traceback (most recent call last):
  File "C:\nxt-python-2.2.2\examples\spin.py", line 3, in <module>
    import nxt.locator
  File "C:\Python27\lib\nxt\__init__.py", line 15, in <module>
    from nxt.locator import find_one_brick, Method
  File "C:\Python27\lib\nxt\locator.py", line 49
    if not silent: print("USB module unavailable, not searching there", file=sys.stderr)
                                                                            ^
SyntaxError: invalid syntax

tcwan, 2016-12-28:
If you're not using the LEGO Fantom drivers, you don't need Pyfantom.
Just define 
'method = usb=True, bluetooth=False, fantomusb=False, fantombt=False' in your config file.

Did you install libusb?
(Sorry I don't use Windows, I don't know the details of where to get the relevant packages)

feinstaub, 2018-04-21:
@guruhj: Did you try to use Python3 lately?

schodet, 2021-11-16:
```
if not silent: print("USB module unavailable, not searching there", file=sys.stderr)
```

This corresponds to the python3 code on the master branch, it will not work with python2.

Here is the corresponding code from the python2 branch:

```
if not silent: print >>sys.stderr, "USB module unavailable, not searching there"
```

closed 2021-11-16 by schodet