schodet/nxt-python #94: Reading from sensors slows down python. Threading doesn't help

marfay, 2016-11-27:
Hey guys,
I have a problem with reading data from all sensors throught NXT-python library, The problem is that it slows down python terribly and even using a special thread for this operation doesn't help at all.
In my Kivy app I am doing interactive display which react with mouse movement. The display is redrawing with 60FPS. But thats not important. The thing is everytime sensor reads data, whole app freezes for like 0.2 secs. I have absolulety no idea whats going on. I am quite hopeless because such thing should be solved by multithreading but it isnt.
Thanks for help, Martin

Code Example of how I read data:
 ```
def sensor_ultrasonic(self):
        self.Touch_sensor=nxt.sensor.Touch(self.brick,nxt.PORT_4)
        threading.Thread(target=self.touch_press).start()
  
def touch_press(self):
     t = self.Touch_sensor
      while True:
           if t.is_pressed()==True:
              print('PRESSED')
           time.sleep(0.5)
```

feinstaub, 2018-04-21:
Hi @marfay: Did you try with the latest master for Python3?

schodet, 2021-11-16:
Tried to reproduce with python3, on the master branch, commit 43072b3, using this program:

```py
import nxt.locator
import time
import sys
import threading

class Test:
    def __init__(self):
        self.brick = nxt.locator.find_one_brick()

    def sensor_ultrasonic(self):
        self.Touch_sensor = nxt.sensor.Touch(self.brick, nxt.PORT_4)
        threading.Thread(target=self.touch_press).start()

    def touch_press(self):
        t = self.Touch_sensor
        while True:
            if t.is_pressed() == True:
                print("PRESSED")
            time.sleep(0.5)

t = Test()
t.sensor_ultrasonic()
while True:
    time.sleep(0.01)
    print(".", end="")
    sys.stdout.flush()
```

There does not seems to be a problem anymore, please reopen with a test program if you can reproduce.

closed 2021-11-16 by schodet