schodet/nxt-python #95: USB Errors on MacOS-Sierra

GolddolphinSKB-Fexcraftian, 2016-12-22:
So I got NXT Pyton and PyUSB downloaded, and they are both installed in python 3 without errors
`import time
import nxt.locator
from nxt.sensor import *
b = nxt.locator.find_one_brick(name = 'Computer')`

The NXT brick connects via USB fine, and is called "Computer'. When I run the code I get multiple errors, mainly from the USB backend.

`Traceback (most recent call last):
  File "/Users/jonathan/Desktop/NXT/latency.py", line 8, in <module>
    b = nxt.locator.find_one_brick(name = 'Computer')
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nxt/locator.py", line 124, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nxt/locator.py", line 45, in find_bricks
    for s in socks:
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nxt/usbsock.py", line 84, in find_bricks
    for bus in usb.busses():
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/usb/legacy.py", line 353, in busses
    sorted(core.find(find_all=True), key=lambda d: d.bus),
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/usb/core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available`

GolddolphinSKB-Fexcraftian, 2016-12-22:
Okay, fixed the backend error. It seems that you need to download libusb, as pyusb calls this, and it isn't there. http://www.dm9.se/?p=659 Seems to work.
Please add this ton the tutorial

GolddolphinSKB-Fexcraftian, 2016-12-22:
New error: 
--------------------------------------------------------------------------------
/Library/Frameworks/Python.framework/Versions/3.5/bin/python3.5 /Users/*********/Desktop/NXT/latency.py
Bluetooth module unavailable, not searching there
No brick was found.
Traceback (most recent call last):
    Is the brick turned on?
  File "/Users/*********/Desktop/NXT/latency.py", line 6, in <module>
    For more diagnosing use the debug=True argument or
    b = nxt.locator.find_one_brick(name = 'Computer')
    try the 'nxt_test' script located in /bin or ~/.local/bin
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/nxt/locator.py", line 154, in find_one_brick
    raise BrickNotFoundError
nxt.locator.BrickNotFoundError
--------------------------------------------------------------------------------

tcwan, 2016-12-28:
NB: I have not used nxt-python with python3, but from my experience with nxt-python running on python2.7 I couldn't get libusb to work in the past.

pyfantom was developed as a wrapper around the Fantom drivers from LEGO. You will need to install the latest Fantom driver (LEGO Mac Driver 1.2.0) and [pyfantom](http://git.ni.fr.eu.org/pyfantom.git). If you're using MacPorts, I have user portfiles (extract and put in $USER/ports/) [python.zip](https://github.com/Eelviny/nxt-python/files/675089/python.zip) to help simplify the process. 

The only issue with using Fantom is that it is 32-bit only.
With MacPorts, you'll have to compile python with +universal, and install the py-objc and py-objc-cocoa packages as +universal as well.

Lastly, you'll need to edit .nxt_python to use the Fantom Drivers.
`[Brick]
name = 
host = 
strict = 0
method = usb=False, bluetooth=False, fantomusb=True, fantombt=False`

To invoke the software, you should start python with something like the following (to force 32-bit mode):
`\#!/bin/sh
\# If using Apple's python, use the following to enable 32-bit python
\# arch -i386 does not work reliably except with specific python versions
\# python2.6 and python2.7 is known to work with 'arch -i386'
\#
\# Apple recommends the use of the following Environment Variable to control
\# 32-bit vs. 64-bit python selection
\# export VERSIONER_PYTHON_PREFER_32_BIT="yes"

DARWIN=\`uname\`
PYTHON=python
PYTHONVER=\`$PYTHON --version\`

APP=<fill in your app path>
if [ $DARWIN == "Darwin" ]; then
        echo "$PYTHONVER Running on Mac OSX Platform!"
        EXP="export VERSIONER_PYTHON_PREFER_32_BIT=yes"
        ARCH="arch -i386"
else
        echo "$PYTHONVER Running on Default Platform!"
        EXP=
        ARCH=
fi
\# Debugging Info
\# echo "dirname:" `dirname $0`
\# echo "basename:" `basename $0`
cd `dirname $0`
\#echo $EXP; $ARCH $PYTHON $APP $*
$EXP; $ARCH $PYTHON $APP $*
`

tcwan, 2016-12-28:
By the way, Fantom Bluetooth support in nxt-python doesn't work reliably for me, I can send commands but not receive any replies.

tcwan, 2016-12-29:
Using MacPorts, with libusb and py27-pyusb installed (in addition to nxt-python).

$ cat ~/.nxt-python 
[Brick]
name = 
host = 
strict = 0
method = usb=True, bluetooth=False, fantomusb=False, fantombt=False

$ python
Python 2.7.13 (default, Dec 23 2016, 05:05:58) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.38)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
\>>> import nxt
\>>> newbrick = nxt
\>>> nxt.find_one_brick()
\<nxt.brick.Brick object at 0x10e4c6d50>

tcwan, 2016-12-29:
[python.zip](https://github.com/Eelviny/nxt-python/files/676646/python.zip)

Revised Portfile for nxt-python for MacPorts. It include py27-pyusb and py27-lightblue as dependencies.

GolddolphinSKB-Fexcraftian, 2016-12-29:
Right, Ive fixed this myself. 10 hours and this is the problem:

info = b.get_device_info() 

is a tuple which includes, for some reason is:
(b'Computer\x00\x00\x00\x00\x00\x00\x00', '00:16:53:18:2A:D1', 0, 91912)

I believe the 'b' is put in there when the unpack() function is called (Can't remember where)
and this errors the

if name and info[0].strip('\x00') != name:

as it cannot strip (Just simple string manipulation stuff), could be as I am running on 3.5.1

a fix is to change the info tuple to a list using list() function, then set the 0th [First] element to the name of the brick, in my case 'Computer'

`            if host and info[1] != host:
                if debug:
                    print("Warning: the brick found does not match the host provided (get_device_info).")
                if strict:
                    s.close()
                    continue
            info = list(info)
            info[0] = 'Computer'
            # /\ Inserted
            print(info[0])
            print(info[0].strip('\x00'))
            if name and info[0].strip('\x00') != name:
`

This is a bodge, and not a permanent fix, please can someone fix the origin of the problem, as I haven't the expertise for this.

Thanks.

GolddolphinSKB-Fexcraftian, 2016-12-30:
Okay, Removed the bodge by converting the bytes to a str, then formatting that way.

>             if host and info[1] != host:
                if debug:
                    print("Warning: the brick found does not match the host provided (get_device_info).")
                if strict:
                    s.close()
                    continue
            info = list(info)
#            info[0] = 'Computer'
            info[0] = str(info[0])
            info[0] = info[0][2:(len(info[0])-1)]
            print('Info[0] is {}'.format(info[0]))
            info[0] = info[0].strip('\\x00')
            print('Info[0] is {}'.format(info[0]))
            # /\ Inserted
            print('Name parsed is {}'.format(name))
            if info[0] != name:
                if debug:
                    print("Warning; the brick found does not match the name provided.")
                if strict:
                    s.close()
                    continue
            return b

>

GolddolphinSKB-Fexcraftian, 2016-12-31:
Just to clear things up, the b is the bytes transmission format, I believe a version change of python
threw things.

Eelviny, 2017-01-04:
@GolddolphinSKB-Fexcraftian, thanks for the bug report - you're correct in the fact that the Python version update thew things... I ran the whole thing through a converter, but I haven't yet resolved the data type mess. I'll merge changes into mainline when I can.

Eelviny, 2017-01-04:
@tcwan, I've taken your Portfile and added it into [this gist](https://gist.github.com/Eelviny/8057aca56d29714df37f0062dd9b58a1) to save for later.

dlech, 2018-04-28:
USB is working for me on macOS with commit a7667d530215a47a462e495c55ad15f249456e21. I'm guessing this issue can be closed?

GolddolphinSKB-Fexcraftian, 2018-04-30:
Yeah, this is fixed.

closed 2018-04-30 by GolddolphinSKB-Fexcraftian