schodet/nxt-python #96: USB Error

avinashlego, 2016-12-27:
I'm using Raspbian on a Raspberry Pi. My python is version 3. My NXT-Python version is 2.2.2. 
My NXT firmware is 1.31. I'm connecting via USB. I have PyUSB installed and the location config file written as well. The error message I'm getting is shown in the screenshot.
[image]

Circuitmage, 2017-04-04:
I have spent about 10 hours porting an NXC program over to Nxt-Python using python 3.x as all the files I installed appeared to be for python 2.x and required heavy corrections to all the NXT files. I also had to add my own Hitechnic sensor function for their PIR sensor, which I have not been able to test yet (as none was part of the their sensor file). Then I find I could not connect to the brick via USB. I spent another 4 hours trying to get that to work, running the usb script, then found an issue here stating  that the USB implementation was only valid for python 2.7. So I installed Python 2.7 on my machine, and now having various issues with path assignments and pack installations. I have to say the NXC package is much simpler . I think I wrote my whole program in about an hour in NXC using Bricx Command Center. The documentation for the C package is about 2000 pages long, with reference to every command module , examples and wide range of capability for the brick (esp display functions, and lots of sensor functions). I really want this to work, but if I cannot get USB communication on my Win7 machine, I will have to abandon ship. If we can at least get the USB to work, I would be happy to share my efforts.

tcwan, 2017-04-12:
I don't know how many people use Windows with libusb. The last time I tried installing libusb on Win7 it wasn't successful. I'm by no means familiar with Win7 and how to configure it, so that may be due to my own unfamiliarity. 

However, I'd suggest using Linux (or MacOS) if you're interested in using nxt-python, the libusb libraries are much easier to install and configure on Linux, for example.

Eelviny, 2018-02-27:
Closing due to inactivity

closed 2018-02-27 by Eelviny