schodet/nxt-python #97: Propose removal of Fantom backend from nxt-python

tcwan, 2016-12-29:
Since the Fantom backend is 32-bit only, and isn't well support in newer releases of MacOS, I'm thinking that is is better to remove it, as I've now discovered that the libusb 1.0.21 release seems to work fine on MacOS.

GolddolphinSKB-Fexcraftian, 2016-12-29:
It does indeed. Fantom is also built for 2.7, I updated the syntax for 3, and you can get the file here.
[https://www.dropbox.com/s/yy5fpbg41w9rlgz/pyfantom.py?dl=0](url)

tcwan, 2017-01-03:
Thanks! From your file, it looks like the only changes needed are to the print statement syntax.
I've not worked with Python 3.x before so I was hesitant on trying to port it.

tcwan, 2017-01-03:
@GolddolphinSKB-Fexcraftian
By the way, were you using Bluetooth or USB to connect to the NXT? The pyfantom.py file you linked had BT enabled but it fails due to incompatible API calls in Sierra.

GolddolphinSKB-Fexcraftian, 2017-01-03:
I used LibUSB Only, and Im pretty sure the file I linked had the print statements fixed.

Eelviny, 2017-01-04:
Hi all,

Sorry for being late with replies, been moving house!
Agreed on removing the Fantom backend, the drivers are now obsolete. 

@GolddolphinSKB-Fexcraftian while the Fantom backend is going to be removed, I'll first create a commit merging your updated code, before removing it. This is so people can go back in history and get the update - for legacy purposes :)

GolddolphinSKB-Fexcraftian, 2018-03-05:
Removed fantom support for the master branch,
Fantomsock branch will be kept with fantom support.

closed 2018-03-05 by GolddolphinSKB-Fexcraftian

tcwan, 2018-03-09:
Just a comment (I'm fine with fantom support being dropped).
I found out that on macOS version of the EV3 Programmer, Fantom is still being used to communicate with the NXT/EV3. Fantom is currently version 1.3.0f1 when installed using the EV3 Programmer Installer.

- NXT support is 32-bit only.
- EV3 support is done using 64-bit (not sure if it has 32-bit support).

So it looks like NXT support will probably be dropped from macOS after 10.13 (High Sierra) since Apple will remove support for 32-bit apps in the future, unless National Instruments update Fantom 64-bit to support NXT.

Eelviny, 2018-03-09:
I'll keep an eye on Fantom - it's always good to be conservative with code removal, so I've ensured that there is a separate branch holding that code if it becomes more useful in the future.

But yes, the preferred method is libusb, and I would prefer it if we supported one method instead of two! Thanks for the info though - I don't have a mac to test with so useful to know.