schodet/nxt-python #98: Errors.

Zoertchen, 2017-01-03:
I'm not this experienced in PC-Things?
Can someone explain, what i've done wrong?
I'm using a MacBook v. 10.9.5

running install
running build
running build_py
running build_scripts
running install_lib
byte-compiling /Library/Python/2.7/site-packages/nxt/bluesock.py to bluesock.pyc
  File "/Library/Python/2.7/site-packages/nxt/bluesock.py", line 55
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

byte-compiling /Library/Python/2.7/site-packages/nxt/brick.py to brick.pyc
  File "/Library/Python/2.7/site-packages/nxt/brick.py", line 215
    class Brick(object, metaclass=_Meta): #TODO: this begs to have explicit methods
                                 ^
SyntaxError: invalid syntax

byte-compiling /Library/Python/2.7/site-packages/nxt/ipsock.py to ipsock.pyc
  File "/Library/Python/2.7/site-packages/nxt/ipsock.py", line 50
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

byte-compiling /Library/Python/2.7/site-packages/nxt/locator.py to locator.pyc
  File "/Library/Python/2.7/site-packages/nxt/locator.py", line 49
    if not silent: print("USB module unavailable, not searching there", file=sys.stderr)
                                                                            ^
SyntaxError: invalid syntax

byte-compiling /Library/Python/2.7/site-packages/nxt/usbsock.py to usbsock.pyc
  File "/Library/Python/2.7/site-packages/nxt/usbsock.py", line 67
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

running install_scripts
changing mode of /usr/local/bin/nxt_push to 755
changing mode of /usr/local/bin/nxt_server to 755
changing mode of /usr/local/bin/nxt_test to 755
running install_egg_info
Removing /Library/Python/2.7/site-packages/nxt_python-3.0-py2.7.egg-info
Writing /Library/Python/2.7/site-packages/nxt_python-3.0-py2.7.egg-info

Thank you

Zoertchen, 2017-01-04:
If i try to run code, using the nxt-package, i get this Error:

Traceback (most recent call last):
  File "/var/folders/t_/_jx20qcj5w503f_qdw6jjs_c0000gq/T/pythonInTerm.DAUaHuwZ", line 1, in <module>
    import nxt.locator
  File "/Library/Python/2.7/site-packages/nxt/__init__.py", line 15, in <module>
    from nxt.locator import find_one_brick, Method
  File "/Library/Python/2.7/site-packages/nxt/locator.py", line 49
    if not silent: print("USB module unavailable, not searching there", file=sys.stderr)
                                                                            ^
SyntaxError: invalid syntax

tcwan, 2017-01-06:
You checked out the Python 3.x branch of nxt-python and tried to run it on Python 2.x.
Python 2.x does not use the print() syntax.

I am not familiar with forward compatibility in Python 2.7, I thought it supported the new 3.x print() syntax?

I'd suggest reading the comments in https://github.com/Eelviny/nxt-python/issues/95
regarding installing nxt-python using MacPorts, it might be easier to get it configured than trying to install it for the system Python Libraries.

Eelviny, 2018-02-27:
Issue solved - running a Python3 build on Python2. Closing

closed 2018-02-27 by Eelviny