schodet/nxt-python #99: Import Error

jbmorris29, 2017-01-10:
I've installed NXT-Python, PyBluez, PyUSB, and Pyfantom, but I keep getting this error when I just try importing nxt.locator:

 ~/Python/nxt-python-2.2.2/scripts $ python nxttry.py 
Traceback (most recent call last):
  File "nxttry.py", line 1, in <module>
    import nxt.locator
ImportError: No module named locator

I get it when I try import nxt.locator, from nxt import locator, and import nxt then import locator.

Any help would be appreciated.

GolddolphinSKB-Fexcraftian, 2017-01-11:
`import nxt.locator`

GolddolphinSKB-Fexcraftian, 2018-03-01:
Have you got a folder called ``nxt`` in the directory? It may be trying to find the module in there.

feinstaub, 2018-04-21:
@jbmorris29: any luck with the hint from GolddolphinSKB-Fexcraftian ?

schodet, 2021-11-16:
No response, closing.

closed 2021-11-16 by schodet