schodet/nxt-python #101: libusb1 module not found error

kk6axq, 2017-01-19:
After installing libusb1, pyfantom, and nxt-python, I get the following error:
`Traceback (most recent call last):
  File "aMove.py", line 5, in <module>
    n = nxt.find_one_brick()
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 112, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 43, in find_bricks
    for s in socks:
  File "/usr/local/lib/python2.7/dist-packages/nxt/usbsock.py", line 84, in find_bricks
    for bus in usb.busses():
  File "/usr/local/lib/python2.7/dist-packages/usb/legacy.py", line 353, in busses
    sorted(core.find(find_all=True), key=lambda d: d.bus),
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1253, in find
    import usb.backend.libusb1 as libusb1
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 39, in <module>
    import usb.libloader
  File "/usr/local/lib/python2.7/dist-packages/usb/libloader.py", line 32, in <module>
    import ctypes.util
  File "/usr/lib/python2.7/ctypes/util.py", line 86, in <module>
    import re, tempfile, errno
  File "/usr/lib/python2.7/tempfile.py", line 35, in <module>
    from random import Random as _Random
  File "/home/lukas/Desktop/NXT/random.py", line 6, in <module>
    n = nxt.find_one_brick()
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 112, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 43, in find_bricks
    for s in socks:
  File "/usr/local/lib/python2.7/dist-packages/nxt/usbsock.py", line 84, in find_bricks
    for bus in usb.busses():
  File "/usr/local/lib/python2.7/dist-packages/usb/legacy.py", line 353, in busses
    sorted(core.find(find_all=True), key=lambda d: d.bus),
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1253, in find
    import usb.backend.libusb1 as libusb1
AttributeError: 'module' object has no attribute 'libusb1'
`
When I first installed the software, it worked, but now it doesn't. I have reinstalled all of the packages with their setup.py files, and it hasn't fixed the issue. Any suggestions?
Thanks,
kk6axq

tcwan, 2017-01-20:
What distro are you using?
I think you probably need pyusb as well.
By the way, you don't need pyfantom if you're on Linux.

kk6axq, 2017-01-20:
I'm using ubuntu 16.04. I've installed pyfantom, pyusb, libusb1, and nxt-python. I tried doing find_one_brick without pyfantom and it didn't work. The weird thing is that I'm able to import usb.backend.libusb1 in the python interpreter, even though that's what's causing the above issue...

kk6axq, 2017-01-22:
I figured out my problem, I had a file called random.py that was being imported when the code called `import random` this caused an import loop as the modules it imported import random, etc. Renaming the file fixed the problem. Thanks for all the help!

closed 2017-01-22 by kk6axq