schodet/nxt-python #102: Invalid Syntax (Most likely due to Terminal Python)

ElPumbledor, 2017-02-22:
Hi all, 

I've got Python3 installed on my Pi & my Laptop but my terminal runs in Python2x when trying to install this. I've tried searching for how to change this but I've been unsuccessful so I'm finally caving to ask for some help.

I changed the `print` parentheses myself but I still have two syntax errors which currently stop the install, I also assume that once I've fixed those I'll get more errors. 

Is there any chance someone can point me in the right direction to install this and fix the terminal running python2x when installing? I wouldn't ask if I hadn't already done all I could. (Feel free to shout obscenities if this is a really easy fix I should of found...)

GolddolphinSKB-Fexcraftian, 2017-02-22:
[image]
To run it as python3, but I guess thats not what you want.

Show the errors here, I will probably be able to help

Eelviny, 2017-10-09:
Closing due to inactivity

closed 2017-10-09 by Eelviny