schodet/nxt-python #103: BrickNotFoundError

Hermlon, 2017-02-27:
Yes, I am sure I installed PyBluez (0.22) correctly. I use nxt-phyton 2.2.2 from the releases page and the firmware version of my nxt is 1.31. When I run any test program (like nxt_test) some seconds nothing happens. Then my brick makes a sound and shows Passkey:1234. After I hit ok the Python log says this: 
`Error while running test:
  File "/home/hermlon/Desktop/nxt-python-2.2.2/scripts/nxt_test", line 16, in <module>
    b = nxt.locator.find_one_brick()
  File "/usr/lib/python2.7/site-packages/nxt/locator.py", line 137, in find_one_brick
    raise BrickNotFoundError`
I do not know what I am doing wrong...

Hermlon, 2017-03-02:
With debug mode enabled there are some more information (I already tried running as root):
`== RESTART: /home/hermlon/Desktop/nxt-python-2.2.2/examples/test_sensors.py ==
Warning: Config file (should be at /home/hermlon/.nxt-python) was not read. Use nxt.locator.make_config() to create a config file.
Host: None Name: None Strict: True
USB: True BT: True Fantom: False FUSB: False FBT: False
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/nxt/locator.py", line 118, in find_one_brick
    b = s.connect()
  File "/usr/lib/python2.7/site-packages/nxt/bluesock.py", line 40, in connect
    sock.connect((self.host, BlueSock.PORT))
  File "<string>", line 5, in connect
BluetoothError: (13, 'Permission denied')
Failed to connect to possible brick

Traceback (most recent call last):
  File "/home/hermlon/Desktop/nxt-python-2.2.2/examples/test_sensors.py", line 6, in <module>
    b = nxt.locator.find_one_brick(debug=True)
  File "/usr/lib/python2.7/site-packages/nxt/locator.py", line 137, in find_one_brick
    raise BrickNotFoundError
BrickNotFoundError
>>> `

RoberWare, 2017-07-24:
Same issue here :(
Someone knows how to repare it?

felipetrentin, 2018-01-06:
same problem here :(

GolddolphinSKB-Fexcraftian, 2018-03-01:
Try running the script as sudo. From the command line:
``sudo python2.2 filename``
Then you may face problems with the the modules not being installed system-wide.

auryan898, 2018-03-02:
Issue #121 might have a solution to this. find_one_brick is tricky, but this solution is pretty reliable.

GolddolphinSKB-Fexcraftian, 2018-03-02:
There was a solution I used to add permission by connecting to it via command line, But I've forgotten how now...

If you type ``BluetoothError: (13, 'Permission denied')`` into google, there are a few stack overflow posts about enabling permissions for a device.

hermlon, 2018-03-25:
According to this https://github.com/pybluez/pybluez/issues/116 you should pair the device first with a Bluetooth manager and then try to connect. It worked for me when I did so. 
(I used this to connect)
`import nxt.bluesock
b = nxt.bluesock.BlueSock('00:16:53:15:BF:19').connect()`