schodet/nxt-python #106: How do I start?

alexbervoets, 2017-03-23:
Hello,

I'm a teacher in a Belgian school. We have a few Lego NXT2.0 robots and I would like to program them with Python. 
Is there some sort of manual where I can find some information on how to get started? I've searched this page but so far it's all very difficult.
How do I program and download the code to the NXT console?

Thank you
Alex

auryan898, 2017-03-24:
Hello Alex,
I have been using nxt-python for quite some time now, and from the looks of things I don't think the developers of the project ever plan on making thorough documentation that tells users what to do, they expect people to figure it all out from examples alone.  I've been using the repo for a while with some Bricks at my school and I can help.

First of all, the code cannot be downloaded to the nxt brick, with nxt-python we make programs that run  on a computer. That computer controls the nxt brick via bluetooth(quite useful) or usb connection(not so useful).  This tethering lets people control the nxt from a distance, like with keyboard commands or a game controller.

Hopefully you've already downloaded nxt-python release 2.2, run the setup.py or install.sh or install.bat file, and you've also installed the other Prerequisites listed in the homepage of the nxt-python repository.
The developers were actually pretty good about installation instructions, but not so much on coding instructions.  I haven't gotten around to making my own documentation yet, but I'll also provide you with examples to start off.
This is my repo: https://github.com/MiltonStatic5060/nxt-comp-app/
I've done some stuff recently and it might help you, but I've also got some confusing stuff because I'm using extra time things and threading and game controllers too. You'll find the nxt-python developer's example codes there too at my repo.  So far I've just been figuring out functions from the source code, which is in the "nxt" folder in the nxt-python release 2.2 zip that you downloaded. If you need extra help with this I'd be glad to help, but maybe one of the developers will answer your questions better (if they ever get around to responding)

Have fun coding!
-Ryan

alexbervoets, 2017-03-24:
Hello Ryan,

thank you very much for your response... It's not easy to find easy documentation, especially when English is your fourth language :(

I'll give it a try. I haven't downloaded all the prerequisites but I'm going to.

I'll be happy if the robot moves ;)
Once I've gotten to that point, I'll figure the rest out.

I'll keep you posted
Alex

alexbervoets, 2017-03-24:
I've installed all the items (I think) but got this error using the USB connection (my laptop has no bluetooth):

USB module unavailable, not searching there
Bluetooth module unavailable, not searching there

Traceback (most recent call last):
  File "C:\Users\Alex\Downloads\nxt-python-2.2.2\nxt-python-2.2.2\examples\spin.py", line 12, in <module>
    b = nxt.locator.find_one_brick()
  File "C:\Python27\lib\site-packages\nxt\locator.py", line 112, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "C:\Python27\lib\site-packages\nxt\locator.py", line 80, in find_bricks
    raise NoBackendError("No selected backends are available! Did you install the comm modules?")
NoBackendError: No selected backends are available! Did you install the comm modules?
>>>

GolddolphinSKB-Fexcraftian, 2017-03-24:
You need to install LibUSB, it is a backend for one of the backends.
If you need more help, Ask, I spent quite a while fixing some problems, and have quite a large knowledge of its structure..

alexbervoets, 2017-03-24:
GolddolphinSKB-Fexcraftian,

thank you, I'm going to try it this monday.

Any information would be welcome...
Greetings
Alex

auryan898, 2017-03-25:
Alex,
I've always had trouble with the usb connection, so instead I use the built in method that just connects to bluetooth. This will need PyUsb installed, bluetooth availability, and the build number of the specific nxt brick you want to connect to.

Here is the code:

import nxt, time
import nxt.bluesock # Make sure you remember this!
b = nxt.bluesock.BlueSock('00:16:53:10:22:3D').connect() #replace the string with the build number from Nxt Version in the settings menu on the Brick

Now with this technique the only issue is that you have to pair to the brick beforehand using the normal bluetooth service on computer before running this bit of code.

Once connected you can use this code for nxt motors:
motorLeft = nxt.Motor(b, nxt.PORT_A)

while 1:
motorLeft.run(50) #nunber from -100 to 100
time.sleep(.1)

These are links to some examples I have. Hopefully they're not too confusing.

https://github.com/MiltonStatic5060/nxt-comp-app/blob/master/examples/spin.py

https://github.com/MiltonStatic5060/nxt-comp-app/blob/master/opmodes/testMotorA.py

GolddolphinSKB-Fexcraftian, 2017-03-26:
The problem of having to connect to it beforehand seems quite bad...
On macOS(OSX) the official NXT software no-longer runs... RIP

auryan898, 2017-03-26:
The connecting thing is only so that the computer and nxt brick recognize each other, this pairing only has to be done once for each new brick. Once paired the first time, the computer and brick don't have to do it again, the program will be able to connect by itself without any other hassle ( at least it should ).

auryan898, 2017-03-26:
Also you didn't say that you were on a macOS. It's important because you actually have to install [LightBlue](http://lightblue.sourceforge.net/) instead according to the nxt-python [Requirements Section](https://github.com/Eelviny/nxt-python)

alexbervoets, 2017-03-26:
auryan898,

I'm running windows 7. But the computers in my classroom aren't equipped with bluetooth. So that's why I was using the USB connection. Tought it would be the easy way. 
If bluetooth should be the easier way, I can always buy those USB bluethooth devices.

alexbervoets, 2017-03-27:
I've downloaded the LibUSB... These files contain an INSTALL-file but I can't run it (no extension). The documentation says I have to put the files in de WIN32 directory. But I don't have such a directory. I have a System32 folder tough... 

I've also tried to use the bluetooth connection: the laptop has found and installed the NXT brick but the brick keeps saying: "line is busy".

But maybe I'm using the wrong program to send my code. Where do I type the code? Python IDLE GUI or the command line? 

Thanks again for the help guys (and the patience ;)   )

auryan898, 2017-03-28:
command line running

python programName.py

always works for me. Line Busy means the nxt thinks it's already connected to something or your bluetooth chip is already connected to something.  Best bet is to turn the nxt on and off and then start your program from command line again.

GolddolphinSKB-Fexcraftian, 2017-04-12:
To install LibUSB, you will have to find an installer, Sadly I can't test it out, as I am running macOS

Eelviny, 2018-02-27:
Old ticket - closing

closed 2018-02-27 by Eelviny