schodet/nxt-python #107: Unstable bluetooth

lukzmu, 2017-03-28:
Hey,

I do have a problem running my nxt-python code on a Macbook Pro. I tried using the internal bluetooth and external dongles, but it seems that each time it looses the connection after some time (usually around 20 iterations).

Here is how the pseudocode looks like:

```
for 0 -> 50:
   get compass reading
   do science stuff
   set motor A and B to desired power
```

It seems like at some point the buffer is overflown with communication (each iteration lasts around 0.1sec) and starts to drop. Running the same code on a Windows laptop worked without problems.

lukzmu, 2017-03-28:
Just tested it over cable and it works perfect, seems like a bluetooth issue on mac os to narrow it down.

auryan898, 2017-03-29:
Well your problem must be rooted in the different bluetooth python package used by nxt-python on macs.  Maybe try altering the source code to use pybluez or some other bluetooth package on macs and seeing if that works

lukzmu, 2017-03-29:
Will modify it in my free time and post a pull request if it works ;)

tcwan, 2017-04-12:
Are you using the Fantom drivers or lightblue?
Also which version of macOS are you on?
I don't think lightblue works on newer macOS versions.

As for Fantom, I've only used the USB support since I could not get bidirectional communications to work properly over Bluetooth (I'm using nxt-python to support remote debugging of ARM bare metal program, which is not the typical use case), so I haven't tried using Bluetooth for normal commands.

I'd be curious to hear about your experience if you can get it to work.
Perhaps if you limit the rate at which you send commands it would be more stable?

Eelviny, 2018-02-27:
No user response - closing

closed 2018-02-27 by Eelviny