schodet/nxt-python #108: Update PyPi version

richin13, 2017-04-11:
I was having a nightmare with nxt-python in Arch Linux because of **Invalid syntax** errors, then I realized that PyPi version of nxt-python is outdated (version 2.2.2 at the time)

Can you @Eelviny please update it to version 3.0?

Thanks

GolddolphinSKB-Fexcraftian, 2017-04-12:
A way to obtain it, is to download the current version from github, then in terminal [Or whatever command line your OS has]

run: cd (Directory of download)
then for python3: python3 setup.py install
if your using python2, write python2, e.t.c.

richin13, 2017-04-12:
Absolutely. That's exactly what I'm doing. 
The issue here is that the PyPi version is outdated so you cannot list nxt-python in the requirements.txt of a project, for example.
In my case, I didn't notice the PyPi version was outdated and I was relying on it, thus the errors and bugs.

Eelviny, 2017-10-09:
Hi, was on holiday for half a year.

I'm going to close this issue as duplicate, as there are many well known problems with the Python3 version that I don't have time to address personally. I have enough time to manage issues and pull requests, which I will accept gladly!

closed 2017-10-09 by Eelviny