schodet/nxt-python #109: NameError: `TemperatureSensor` is not defined

vishen89, 2017-05-04:
Hello all,
I am very new to LEGO brick and Python. I tested the inbuilt examples with spin.py and test_sensors.py and it works great. 
But when i removed one sensor and attached the temperature sensor (as shown in the code below) it gives the **Name Error**.
I included the code that was provided here [https://github.com/Eelviny/nxt-python/wiki/Digital-Sensor-Guide]  in generic.py 

`import nxt.locator
from nxt.sensor import *

b = nxt.locator.find_one_brick()

print 'Touch:', Touch(b, PORT_1).get_sample()
print 'Temperature:', TemperatureSensor(b, PORT_2).get_sample()
print 'Light:', Light(b, PORT_3).get_sample()
print 'Ultrasonic:', Ultrasonic(b, PORT_4).get_sample()`

`Touch: False
Light: 0
Ultrasonic: 24
Temperature:

Traceback (most recent call last):
  File "/home/pi/Desktop/nxt-python/nxt-python-2.2.2/examples/test_sensors.py", line 12, in <module>
    print 'Temperature:', TemperatureSensor(b, PORT_2).get_sample()
NameError: name 'TemperatureSensor' is not defined`

Can someone help me as to why am I getting this error? or why the temperature sensor is not being detected?

Regards
Vishvesh

auryan898, 2017-05-04:
Temperature Sensor has no name for it defined in the source code of nxt-python.
Sensors for the nxt are either analog or digital. If the temperature sensor is somehow just analog, then it could give you back information if you write:

`print 'Temperature:', Light(b, PORT_2).get_sample()`

or try

`print 'Temperature:', Sound(b, PORT_2).get_sample()`

or try 

`print 'Temperature:', Touch(b, PORT_2).get_sample()`

I would try all of these different lines to see if any of them work.

vishen89, 2017-05-04:
I tried as per your suggestion but there was no behavior from the temperature sensor. And as per the data from LEGO website, the sensor is Digital. I expected it to be anyways. I know I have to use the i2c address and requires a lot of coding, but I thought the code provided from the link mentioned above should suffice. 
Also I am a newbie at i2c address.

auryan898, 2017-05-04:
The code provided from the link could work if it follows the right specifications. I couldn't find the specs for an NXT sensor on the internet.  Do you happen to have the specs for your temperature sensor? any list of i2c addresses?

auryan898, 2017-05-04:
http://www.hitechnic.com/blog/wp-content/uploads/HiTechnic-Motor-Controller-Specification.pdf

i2c specs like page 5 of this ^

vishen89, 2017-05-05:
I have been searching the web for the specifications. But no use. 
I just want to understand if my error 
`NameError: name 'TemperatureSensor' is not defined`
is due to "incorrect "I2C Address or I need to include some code somewhere in Digital.py or generic.py 
I have been trying to trace back the steps included for Ultrasonic sensor(digital). But so far I havent been able to run my "simple" code.

Thank you for your time on this @auryan898

auryan898, 2017-05-06:
Ok I think I know what might be going on.  You changed the build code in nxt-python 2.2.2 whatever version and then you re-installed it using the setup.py script? good?
If you followed the code from Ultrasonic as an example then it should work. If you did these instructions and got this error, then show what your new digital.py looks like

vishen89, 2017-05-11:
Sorry for the delayed response. I was busy with another task.
Yes, NXT Python version is 2.2.2

Here is the code that I have included in the generic.py file. There is no change made in the digital.py except for the one line in definition __Init__
 
`def __init__(self, brick, port, check_compatible=True):
        """Creates a BaseDigitalSensor. If check_compatible is True, queries
        the sensor for its name, and if a wrong sensor class was used, prints
        a warning.
        """
        super(BaseDigitalSensor, self).__init__(brick, port)
        **self.set_input_mode(Type.TEMPERATURE, Mode.RAW)`**

generic.py file
`class TemperatureSensor(BaseDigitalSensor):
    """Object for Temperature sensors"""
    I2C_ADDRESS = BaseDigitalSensor.I2C_ADDRESS.copy()
    I2C_ADDRESS.update({'temperature': (0x99, 'h')})

    def __init__(self, brick, port, check_compatible=False):
        if check_compatible:
            print "sorry, it's not supported yet"
        super(TemperatureSensor, self).__init__(brick, port, False)

    def get_temperature(self):
        """Returns temperature in Celsius."""
        # return self.read_value('temperature') # WRONG! returns (23,)
        return self.read_value('temperature')[0] # correct, returns 23
    
    get_sample = get_temperature

TemperatureSensor.add_compatible_sensor(None, 'LEGO', 'Temp. ')`

The I2C address for the temperature sensor I took it from [http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html#table-ofi2c-addresses](url)

Brick: NXT not EV3.

vishen89, 2017-05-18:
`import nxt.locator
from nxt.sensor import *
b = nxt.locator.find_one_brick()
s = nxt.BaseDigitalSensor(b, PORT_1)
print s.get_sensor_info()
`
also when I used this code, to check if I could get the info related to the respective sensor,
I got this for Ultrasonic Sensor.

`Version: `V1.0`
0x56, 0x31, 0x2e, 0x30, 
Product ID: `LEGO`
0x4c, 0x45, 0x47, 0x4f, 
Type: `Sonar`
0x53, 0x6f, 0x6e, 0x61, 0x72, `

But the same when I tried to get the sensor info for Temperature sensor. here is the output.

`Traceback (most recent call last):
  File "/home/pi/Desktop/nxt-python-examples_python2/sensor_info.py", line 8, in <module>
    print s.get_sensor_info()
  File "/usr/local/lib/python2.7/dist-packages/nxt/sensor/digital.py", line 155, in get_sensor_info
    version = self.read_value('version')[0].split('\0')[0]
  File "/usr/local/lib/python2.7/dist-packages/nxt/sensor/digital.py", line 142, in read_value
    raise I2CError, "read_value timeout"
I2CError: read_value timeout
`
What I understand is irrespective of the code that has been included in the **generic.py** for the temperature sensor, the get_sensor_info() should give the data(version, product, type), when the sensor is connected. Am I right??
If true, then I can understand why the **TemperatureSensor** is having the **NameError**, since it is not at all being detected in the first place. 
Hope this additional information helps, so that someone can help me with this.

Vishvesh

auryan898, 2017-05-18:
Yeah I guess if the Temperature Sensor isn't even being detected then you won't get anything back regardless of whatever code you write.

vishen89, 2017-05-22:
Got it resolved the **NameError**. The problem was I had changed the code in the 3.x version and I was running 2.x .Anyways my bad. But now I have a new error which has to do specifically with the I2C address. But I will raise a new issue. Since this issue is now resolved.
Thank you for your time and your help.

commit ae449f86 (2018-04-28): Add class for LEGO NXT Temperature sensor

This adds a new class for the official LEGO NXT Temperature sensor.

Fixes #109
Fixes #111
closed 2018-04-28 by GolddolphinSKB-Fexcraftian