schodet/nxt-python #110: USB BrickNotFoundError (python 2 and 3, nxt 2.0)

Sheogorath2, 2017-05-14:
Firstly I tried to install nxt-python for python3, but when i tried to launch my script I was getting BrickNotFoundError. Enabling debug mode showed that I have insufficient permissions. Trying running the script with sudo showed that the name parsed is None and the same error. 
Then I tried the 2nd python. Running the script as user showed the same permission error. Running it with sudo worked fine but still I want to use the 3rd python and run as user. Did someone encountered such problem?

The nxt-python is installed using pip

Error messages:

[python3 if I run as user](https://gist.github.com/Sheogorath2/6d4f7aa7354b00b0451fa824eba5dd58/raw/e074fff03d8ab3b2318038023e4af072d910dcc5/python3%2520if%2520I%2520run%2520as%2520user)
[python3 if I run as root](https://gist.github.com/Sheogorath2/6d4f7aa7354b00b0451fa824eba5dd58/raw/e074fff03d8ab3b2318038023e4af072d910dcc5/python3%2520if%2520I%2520run%2520as%2520root)
[python2 if I run as user](https://gist.github.com/Sheogorath2/6d4f7aa7354b00b0451fa824eba5dd58/raw/e074fff03d8ab3b2318038023e4af072d910dcc5/python2%2520if%2520I%2520run%2520as%2520user)
[python2 if I run as root (works)](https://gist.github.com/Sheogorath2/6d4f7aa7354b00b0451fa824eba5dd58/raw/e074fff03d8ab3b2318038023e4af072d910dcc5/python2%2520if%2520I%2520run%2520as%2520root%2520(works))

tcwan, 2017-06-02:
I presume that you're on Linux. Did you setup permissions on the USB port for user access?
Sorry I don't remember the steps involved, you should search for udev rules configuration for USB port for detecting Mindstorms device on connect.

Sheogorath2, 2017-06-04:
Thank you for your help. Following [this tutorial](http://bricxcc.sourceforge.net/nbc/doc/nxtlinux.txt) I was able to get script running as user. So the permission problem is solved.

But still if I run the program using python3 it says 

```
Info[0] is mpute
Info[0] is mpute
Name parsed is None
Warning; the brick found does not match the name provided.
Bluetooth module unavailable, not searching there
```

tcwan, 2017-06-29:
Sorry, I don't know about Python3 as I'm not using it.
Sounds like some of the string parsing functions are not working the same way in Python3.
You might need to check if the underlying libraries are Python3 compatible.

GolddolphinSKB-Fexcraftian, 2017-06-29:
Yup, I found that a while ago, https://github.com/Eelviny/nxt-python/issues/95
I haven't put my code into a pull request, because its just a temporary fix.

FlorianPoot, 2017-08-27:
Hello, I'm using Python 3 and the master branch of nxt-python and I just found why Info[0] is mpute. This is because in file locator.py line 139 there is info[0] = "Computer". I don't know why it's there... so just comment it ! Also in file usbsock.py line 68 the print function create a exception so comment it too !

It should be commited but I'm new to GitHub and I don't really know how to do this.

I hope this help and that it will work (It worked for me)

fiddybux, 2017-08-27:
Thanks for this. I gave up on Py3 because of issues like this (so I use Py2
and haven't had this brick not found error). I'm new to Py in general, so
it's been quite a ride so far. Thanks for your contribution!

GolddolphinSKB-Fexcraftian, 2018-03-01:
@FlorianPoot That was from my changes a few months ago, I forgot to remove some of my debugging code.
The newest master branch should work.

dlech, 2018-04-28:
It sounds like the original permission problem would be solved with a udev rule such as the one in the [wiki](https://github.com/Eelviny/nxt-python/wiki/Installation#linux).

GolddolphinSKB-Fexcraftian, 2018-04-28:
Yup.

closed 2018-04-28 by GolddolphinSKB-Fexcraftian

sirdrakeistda, 2022-03-11:
> I presume that you're on Linux. Did you setup permissions on the USB port for user access? Sorry I don't remember the steps involved, you should search for udev rules configuration for USB port for detecting Mindstorms device on connect.
Maybe it would be helpful to add this to the installation (user must be part of group plugdev):
```
sudo echo 'SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0002", GROUP="plugdev", MODE="0777"' >> /etc/udev/rules.d/50-myusb.rules
sudo udevadm control --reload
sudo udevadm trigger
```