schodet/nxt-python #111: Lego Temperature sensor I2C address errors

vishen89, 2017-05-22:
Hi again,
In the previous issue [109](https://github.com/Eelviny/nxt-python/issues/109), I was able to resolve the issue of NameError.
But now I am facing this new error as below.
```
Temperature:

Traceback (most recent call last):
  File "/home/pi/Desktop/Vish/nxt-python-examples_python2/temp_sensor.py", line 8, in <module>
    print 'Temperature:', TemperatureSensor(b, PORT_1).get_sample()
  File "/usr/local/lib/python2.7/dist-packages/nxt/sensor/generic.py", line 88, in get_thermometer
    return self.read_value('temperature')[0] # correct, returns 23
  File "/usr/local/lib/python2.7/dist-packages/nxt/sensor/digital.py", line 142, in read_value
    raise I2CError, "read_value timeout"
I2CError: read_value timeout
```
the code is just this:
```
import nxt.locator
from nxt.sensor import *
b = nxt.locator.find_one_brick()
print 'Temperature:', TemperatureSensor(b, PORT_1).get_sample()
```
I know that this has to do with the **I2C address**, but how do I resolve it. I have no clue neither I have any experience with python. 
I tried to find the specifications for this sensor, but of no use. 
I know this and the earlier issue are related, but I thought I should create a new issue since I got the NameError resolved. 
Thank you for any help in advance.

vishen89, 2017-05-30:
Has anyone been able to get the output from using the Temperature Sensor??
Has anyone worked on it?

vishen89, 2017-06-02:
@Eelviny  Can you help me with this?

Eelviny, 2017-10-08:
Hi there, sorry, I was on vacation for half a year... While I doubt you're still trying to solve this, for sake of completeness, I'll reply anyway.

You're most likely right that this is a problem with the I2C address, but I don't own that sensor and can't confirm the issue. Without the relevant documentation, I'm not sure what I could do other than brute forcing every address there is... don't really have time for that.

I'll leave the issue open, as it is unsolved at the moment. If anyone who owns the sensor could get to the bottom of this, I'll accept that pull request in seconds.

commit ae449f86 (2018-04-28): Add class for LEGO NXT Temperature sensor

This adds a new class for the official LEGO NXT Temperature sensor.

Fixes #109
Fixes #111
closed 2018-04-28 by GolddolphinSKB-Fexcraftian