schodet/nxt-python #112: Issue with passkey when connecting via bluetooth

PullJosh, 2017-07-11:
*Ubuntu 16.10, Python 2.7.12+*

I have a python program running the following lines:
```python
import nxt.bluesock
# ...
b = nxt.bluesock.BlueSock('00:16:53:10:01:F1').connect()
```
It *almost* works. When the code is run, my NXT plays a sound and asks for a passkey. Unfortunately, no matter what passkey I enter, I recieve the following error:
```
Traceback (most recent call last):
  File "Snap-NXT.py", line 131, in <module>
    b = nxt.bluesock.BlueSock('00:16:53:10:01:F1').connect()
  File "/usr/local/lib/python2.7/dist-packages/nxt/bluesock.py", line 40, in connect
    sock.connect((self.host, BlueSock.PORT))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: (52, 'Invalid exchange')
```

How do I resolve this?

auryan898, 2017-07-12:
I had this same problem before. So this isn't quite a problem with nxt-python as it is a problem with pairing the nxt brick first.  Try to connect to the brick via bluetooth prior to running your code.  Once the brick has connected to the computer once then you don't have to pair it again afterwards.  You might have to use another bluetooth software to pair the brick than the one already on your computer.  I installed blueman and that helped.

auryan898, 2017-07-12:
If you are ok with using some command line tools then here's an easy way to pair the brick.
Open a terminal and type in the following

`bluetoothctl`

You will then enter in some commands to ensure the bluetooth transmitter is on on your computer:

`power on`
`agent on`
`default-agent`
`scan on`

Then you will see a list of devices to pair to. (Make sure your nxt brick is on)
Then use the following command:

`pair FF:EE:DD:CC:BB:AA` use the mac address of your brick

Then input a pin code on your computer, and then do the same pin on your brick.
You're computer and brick are now paired and your program should work

PullJosh, 2017-07-12:
That worked perfectly! Thanks a million. :smile:

closed 2017-07-12 by PullJosh