schodet/nxt-python #113: Modules can only be called from main directory?

fiddybux, 2017-08-10:
Is this normal? I managed to get 2.2.2 working with Py 2.4 over USB on Win32, but in order to execute the spin.py example to work I must execute from within the main unzip directory. I guess this is because it calls import nxt.module_name, and the path fails if trying to call from elsewhere, e.g. inside the examples folder. Is this normal expected behaviour? Thanks.

auryan898, 2017-08-11:
Did you install it? Did you run this:
`python setup.py install`

Once you do it, your spin.py should work in any directory

fiddybux, 2017-08-17:
Hi,

Yes, pretty sure it's all installed. I can call 'import nxt' from the py2 console, and no errors appear.

I can run 'spin.py' as long as I copy it to the nxt-python-2.2.2 extract path on my disk. As long as I'm connected to Port B on the brick, the motor spins in one direction and then wobbles back and forth momentarily. Is this the expected behaviour from the 'spin.py' example?

If I attempt to run 'spin.py' from the examples folder, it fails (this is the problem to which I refer in my first post).

Re-running 'python setup.py install' (for me 'py -2 setup.py install') produces the following output:
```
C:\Users\Me\Downloads\Desktop\Python\NXT Python\nxt-python-2.2.2>py -2 setup.py install
running install
running build
running build_py
running build_scripts
running install_lib
byte-compiling C:\Python27\Lib\site-packages\nxt\bluesock.py to bluesock.pyc
  File "C:\Python27\Lib\site-packages\nxt\bluesock.py", line 55
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

byte-compiling C:\Python27\Lib\site-packages\nxt\brick.py to brick.pyc
  File "C:\Python27\Lib\site-packages\nxt\brick.py", line 215
    class Brick(object, metaclass=_Meta): #TODO: this begs to have explicit methods
                                 ^
SyntaxError: invalid syntax

byte-compiling C:\Python27\Lib\site-packages\nxt\ipsock.py to ipsock.pyc
  File "C:\Python27\Lib\site-packages\nxt\ipsock.py", line 50
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

byte-compiling C:\Python27\Lib\site-packages\nxt\locator.py to locator.pyc
  File "C:\Python27\Lib\site-packages\nxt\locator.py", line 49
    if not silent: print("USB module unavailable, not searching there", file=sys.stderr)
                                                                            ^
SyntaxError: invalid syntax

byte-compiling C:\Python27\Lib\site-packages\nxt\usbsock.py to usbsock.pyc
  File "C:\Python27\Lib\site-packages\nxt\usbsock.py", line 67
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

running install_scripts
running install_egg_info
Removing C:\Python27\Lib\site-packages\nxt_python-2.2.2-py2.7.egg-info
Writing C:\Python27\Lib\site-packages\nxt_python-2.2.2-py2.7.egg-info
```
There appears to be a lot of errors, but some are easily explained (e.g. I don't have 'bluesock' installed, for example.) Others, I can't explain at all - they don't mean much to me right now.

In my environment variables (Windows 7), I've made sure Python27 is before Python36-32. So running 'python' from CMD will invoke Py2, rather than Py3. In any case, I always prefer to specify with 'py -2' or 'py -3'.

My C:\Python27\Scripts folder contains these files:
```
nxt_filer
nxt_push
nxt_server
nxt_test
```
Should I be adding the '\nxt-python-2.2.2\nxt folder' to my PATH in order to get scripts to run from anywhere on my system? That's all I can think of to resolve this.

Thanks.

auryan898, 2017-08-17:
I'm on a linux man, so path names will look weird to you, but what you have up there doesn't look right at all.

    running install
    running build
    running build_py
    running build_scripts
    running install_lib
    running install_scripts
    copying build/scripts-2.7/nxt_filer -> /usr/local/bin
    changing mode of /usr/local/bin/nxt_server to 755
    changing mode of /usr/local/bin/nxt_filer to 775
    changing mode of /usr/local/bin/nxt_push to 755
    changing mode of /usr/local/bin/nxt_test to 755
    running install_egg_info
    Removing /usr/local/lib/python2.7/dist-packages/nxt_python-2.2.2.egg-info
    Writing /usr/local/lib/python2.7/dist-packages/nxt_python-2.2.2.egg-info

I don' t like those errors, there on important scripts like brick.py

Post some output of an error occurring from running your spin.py outside of the nxt folder.

(Note: Those scripts are actually useful for other stuff and don't affect your spin.py scripts you can try running them if you want. Go to the scripts folder and try `nxt-filer` or `.\nxt-filer` idk for sure what works for you)

fiddybux, 2017-08-17:
I'm pretty sure I ran (double-clicked) the install.bat again today as well. The problem with that of course is that the cmd window just flashes up, so I ran it from within the cmd (by hand, so to speak), and yes I got the same errors as above.

I'm running by 'buntu box right now, so I can't post an error. Off the top of my head, I recall it mentioning something about the path being wrong, but my feeble human memory may be letting me down or convincing me otherwise. I'll check when I next have the opportunity!

I managed to edit spin.py today to run forwards and backwards at different speeds; so something is working as it should it seems!

auryan898, 2017-08-18:
Ok so I look at those errors you had, and the errors involve code that doesn't look like it should be there...

     File "C:\Python27\Lib\site-packages\nxt\bluesock.py", line 55
        print('Send:', end=' ')

I don't know for sure, I'm just taking a wild guess but just try redownloading release 2.2.2 and just reinstall the same way you have been.

fiddybux, 2017-08-19:
Should I use the master version perhaps? I tried that the first time (I didn't read properly as I'm sure it was unstable), but perhaps you know differently.

auryan898, 2017-08-19:
[https://github.com/Eelviny/nxt-python/releases](https://github.com/Eelviny/nxt-python/releases)

This is what I meant by release 2.2.2, download that as a zip and then install that.  Yes the master branch is likely to be unstable and has those errors that stumped your installation

fiddybux, 2017-08-20:
I am able to get spin.py to work in any direction and speed, despite those
errors.

closed 2017-10-08 by fiddybux