schodet/nxt-python #114: Error while running sudo python setup.py install.

xXRoboBoyXx, 2017-08-22:
I Just ran this and after hours and hours of searching, i find nothing. The error is this:

pi@raspberrypi:~/Desktop/Nxt/nxt-python-master $ sudo python setup.py install
running install
running build
running build_py
running build_scripts
running install_lib
byte-compiling /usr/local/lib/python2.7/dist-packages/nxt/locator.py to locator.pyc
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 49
    if not silent: print("USB module unavailable, not searching there", file=sys.stderr)
                                                                            ^
SyntaxError: invalid syntax

byte-compiling /usr/local/lib/python2.7/dist-packages/nxt/ipsock.py to ipsock.pyc
  File "/usr/local/lib/python2.7/dist-packages/nxt/ipsock.py", line 50
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

byte-compiling /usr/local/lib/python2.7/dist-packages/nxt/bluesock.py to bluesock.pyc
  File "/usr/local/lib/python2.7/dist-packages/nxt/bluesock.py", line 55
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

byte-compiling /usr/local/lib/python2.7/dist-packages/nxt/brick.py to brick.pyc
  File "/usr/local/lib/python2.7/dist-packages/nxt/brick.py", line 215
    class Brick(object, metaclass=_Meta): #TODO: this begs to have explicit methods
                                 ^
SyntaxError: invalid syntax

byte-compiling /usr/local/lib/python2.7/dist-packages/nxt/usbsock.py to usbsock.pyc
  File "/usr/local/lib/python2.7/dist-packages/nxt/usbsock.py", line 67
    print('Send:', end=' ')
                      ^
SyntaxError: invalid syntax

running install_scripts
copying build/scripts-2.7/nxt_test -> /usr/local/bin
changing mode of /usr/local/bin/nxt_push to 755
changing mode of /usr/local/bin/nxt_server to 755
changing mode of /usr/local/bin/nxt_test to 755
running install_egg_info
Removing /usr/local/lib/python2.7/dist-packages/nxt_python-3.0.egg-info
Writing /usr/local/lib/python2.7/dist-packages/nxt_python-3.0.egg-info
Is there something i'm doing wrong...?

auryan898, 2017-08-22:
Is it imperative that you be using nxt-python with python 3.0?  Python 3.0 support was never finished so the last stable release was in python 2.6 or higher.

Eelviny, 2017-09-29:
Hey there,

As @auryan898 has mentioned, the Python3 implementation is not complete. I don't know what distro you're using, but you most likely want to run this instead: `sudo python*2* setup.py install`

closed 2017-10-08 by Eelviny