schodet/nxt-python #117: How to get tacho info

pmu5757, 2017-10-07:
Hello,
It's surely because I'm not good at writing python code, but I don't know how to get the tacho value, I mean a number, not an object.
I tried this but it doesn't work
I have an error with the last line : TypeError: 'int' object is not callable

mc = Motor(b, PORT_C)
mc.reset_position(True)
tacho = mc.get_tacho()
print tacho
print tacho.tacho_count()

Thank you for the help you can give to a noob :-)
Pascal.

auryan898, 2017-10-08:
import nxt
    from nxt.motor import *
    import time
    b = nxt.locator.find_one_brick()
    mc = Motor(b, PORT_C)
    mc.reset_position(True)
    position=int(mc.get_tacho().block_tacho_count)
    mc.run(40,True)
    while position<360:
        position=int(mc.get_tacho().block_tacho_count)
        print "position",position
    mc.brake()
    time.sleep(0.5)
    mc.idle()

I'm going to guess this is what your code looks like?

So as is, your code is not flawed, and it's most likely that this is an issue regarding hardware, and not software.  Eelviny even included comments in some methods saying that low tacho counts aren't reliable.

Nxt motors were never the most precise, so tacho counts are always bound to be off by a lot.  

The type of connection between your computer and the nxt is also important.  Bluetooth makes for a slow and weak connection, so information could be delayed and show jumps in values for ANYTHING.  Maybe you're not using bluetooth and you figured out usb, great, but, each iteration of a while loop doesn't run for the same amount of time. it could be 2 microseconds between loops or 2 seconds, so you'll be getting information at different points in time and the tacho just seems to jump.

auryan898, 2017-10-08:
turn(self, power, tacho_units, brake=True, timeout=1, emulate=True):
        """Use this to turn a motor. The motor will not stop until it turns the
        desired distance. Accuracy is much better over a USB connection than
        with bluetooth...
        power is a value between -127 and 128 (an absolute value greater than
                 64 is recommended)
        tacho_units is the number of degrees to turn the motor. values smaller
                 than 50 are not recommended and may have strange results.
        brake is whether or not to hold the motor after the function exits
                 (either by reaching the distance or throwing an exception).
        timeout is the number of seconds after which a BlockedException is
                 raised if the motor doesn't turn
        emulate is a boolean value. If set to False, the motor is aware of the
                 tacho limit. If True, a run() function equivalent is used.
                 Warning: motors remember their positions and not using emulate
                 may lead to strange behavior, especially with synced motors
        """

pmu5757, 2017-10-08:
Hello,
Thank you for being interested in my question.
Excuse me for the indentation in the code.
I use USB.
I think you're right when you say it's a hardware problem.
I even think it's a problem of communication between PC and NXT. If this communication stops for some millisec, it explains the jumps.
I tried the same kind of code in NXC language ans it works perfectly.
I wanted to use nxt-python because NXC doesn't accept recursivity (my robot is the hanoi towers).
But now I maybe see another solution.
Is it possible with nxt-python to call a program or better part of a program written in nxc and previously uploaded to the nxt ?
In my mind, recusrvivity would be done by Python and precise movements by NXT
Thank you for your help.
Pascal.

Eelviny, 2017-10-08:
Hi @pmu5757,

Yes, it is possible to create a sort of hybrid NXC/NXT-Python build, and I always recommend local processing on the NXT brick for high precision motor control. The problem is latency - a by the time the stop command comes through the bluetooth link to the motor, the motor has overshot its mark.

While I don't have time to figure out an exact solution for you, I can give you some pointers: 

Create an NXC program, that you start manually on the NXT brick. Create a set of functions that continuously listen on certain bluetooth sockets for a message. When the message and parameters are received, execute the function locally. In this case, a message with the amount of units to move the motor.

On the NXT-Python side, investigate the `message_write` function in the `direct.py` file - you want to send bluetooth messages directly to the brick using the backend. This code was written so long ago that I've forgotten how it was all set up, so please write back your findings!

pmu5757, 2017-10-09:
Hi,
I see what you mean but I'm afraid not to be a "EXTREMELY ADVANCED USER" as mentioned in direct.py file. I've no idea how to work with that.
I understand that you don't have time to help me in detail.
So I'll try another option : write a non-recursive program entirely in NXC.
I think I can do that.
Thank you,
Pascal.

Eelviny, 2017-10-11:
Fake it 'till you make it 😄 
The point of NXT-Python is to allow you to hook into other Python libraries, and use more powerful functions such as internet access, other systems, etc. If you don't need all of that, then NXC is the right way for you!

closed 2017-10-11 by Eelviny

pmu5757, 2017-10-19:
Hi,
Here is the result only with NXC :  [https://youtu.be/hnqFwpz3RSs]
You were right : hi precision like that.
Pascal.