schodet/nxt-python #118: New NXT File Management Tool

tcwan, 2017-11-16:
I've created a new tool, nxtfilemgr to replace nxt_filer which is not working properly,  as the NeXT Tool by John Hansen (bricxcc) no longer works on macOS High Sierra.

I'm not sure what is the best way to push the file (I have not used git's push functions before), but you can access it from my repository:
[nxtfilemgr 1.0](https://nxos-armdebug/scripts/nxtfilemgr )

The release version (1.0) has the following commit ID:
[nxtfilemgr 1.0 commit ID](https://github.com/tcwan/nxos-armdebug/commit/9d6d13591ebce443af3c20b44f88588a5081fa07)

This was tested with python2.7 on macOS High Sierra. It is a Gtk+ 3 app.

auryan898, 2017-11-26:
Could you try the tool using python 3? This repo was last being written in python 3 so anything you push to this should be able to work with python 3 too

tcwan, 2017-11-29:
@auryan898 

I've tried to do some cleanup, but I don't have nxt-python for python3 installed, and I'm not sure how to build a package specifically for python3 under MacPorts for macOS.

[nxtfilemgr 1.02 Python 3 cleanup commit ID](https://github.com/tcwan/nxos-armdebug/commit/3a45752bbf3c83a91727dd76675de39401728e80)

tcwan, 2017-11-29:
There are also some missing packages e.g., py-lightblue which I don't know whether there are alternatives.

auryan898, 2017-11-29:
Well that's okay. I only ask if you can use it with python 3 because if you were to push this script to the nxt-python repo right now, it would be best if it were python 3 compatible.  The other option is to push your script to a python 2 part of this repo

tcwan, 2017-11-30:
As I don't have a working python3 configuration for nxt-python currently I'm not confident that everything would be fine. However, I've tried to remove any python2 specific code as much as possible. 

Another thing is I'm not sure how to push the script, as I haven't cloned my own nxt-python repository as I'm not actively working on it. If you're ok with grabbing it from my repository linked above please go ahead. Otherwise I'd need pointers on how to push from a different repository to this one.

tcwan, 2017-12-13:
I've found out that I can upload the script via the web interface to my fork, so I've generated a pull request for it.

closed 2018-02-27 by Eelviny