schodet/nxt-python #119: RCX rotation sensor

pmu5757, 2017-11-20:
Hellow,
I connected and old RCX rotation sensor to my NXT with an adaptation cable.
With NXC, I know how to use it, but not with nxt-python.
Thank you for your help.
Best regards,
Pascal.

pmu5757, 2017-11-20:
Hello, I found the solution : I created a new sensor class called "Angle" using Type.ANGLE, Mode.ANGLE_STEPS
Here is my code if it could interest someone.

class Angle(BaseAnalogSensor):

    def __init__(self, brick, port):
        super(Angle, self).__init__(brick, port)
        self.set_input_mode(Type.ANGLE, Mode.ANGLE_STEPS) 

    def get_angle(self):
        return self.get_input_values().scaled_value	  
 
    get_sample = get_angle

auryan898, 2017-11-29:
which file did you add this code to?

pmu5757, 2018-01-20:
Hello,
I wrote it in my own code, not in any library.
Best regards,
Pascal.

Eelviny, 2018-02-27:
Going to mark this as enhancement, but close this issue - if I feel like it in the future, I'll dig it up and put it in mainline.

closed 2018-02-27 by Eelviny