schodet/nxt-python #121: Connecting via bluetooth. Running ubuntu

HaginCodes, 2018-01-01:
I have tried to connect to the robot via bluetooth when I run the command 
`>>> nxt.locator.find_one_brick()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 137, in find_one_brick
    raise BrickNotFoundError
nxt.locator.BrickNotFoundError
`

It prompts my NXT brick with the passcode 1234 and when I press the check mark it doesn't connect. I also tried doing it vice versa and connecting from my NXT to ubuntu and it didn't connect it would recognize and I would try to connect with all lines but it said "line is busy when ever tried". I wasn't connected to anything on my computer and all connections were available. Could I have some help in trying to connect the two? Thanks in advance.

auryan898, 2018-01-03:
you have to pair the computer and the nxt brick with the normal bluetooth UI first.
Then run your python program

HaginCodes, 2018-01-03:
I have tried that before. I have `blueman` installed and I find the device I request it to connect. On the NXT it receives the request and asks for me to set the code to send over and it is 1234 on the NXT. When I try that and press the check on the NXT the connection drops and it doesn't connect. I would expect a  gui to show up on my computer via blueman saying NXT wants to connect like it does when I connect other devices but it doesn't. I also have tried doing this in terminal using `hcitool scan` and follow the same procedure and `sudo l2ping <bluetooth address>` (or something like that). It prompts for the password on the NXT I accept the passcode and then it fails and stops in the terminal.

auryan898, 2018-01-03:
So I believe this was from a past issue, but I can't seem to remember where it came from.  You need to open a terminal and type in `bluetoothctl`

Then proceed to perform the following commands like `power on`  `agent on` and `scan on`

Once you get that Bluetooth MAC address from the scanning, copy and paste it into `pair _:_:_:_:_ `

```
[bluetooth]# power on
[bluetooth]# agent on
[bluetooth]# default-agent 
[bluetooth]# scan on

Discovery started
[CHG] Controller AA:BB:CC:DD:EE:FF Discovering: yes
[NEW] Device FF:EE:DD:CC:BB:AA Someone's Keyboard

[bluetooth]# pair FF:EE:DD:CC:BB:AA 

Attempting to pair with FF:EE:DD:CC:BB:AA 
[CHG] Device C8:E0:EB:04:52:55 Connected: yes

[agent] Enter PIN code: 1234

[CHG] Device FF:EE:DD:CC:BB:AA Paired: yes

[bluetooth]# exit
```

HaginCodes, 2018-01-04:
That did work I am successfully connected but when I run a script it says otherwise. When I run this simple script it doesn't seem to find it. 

This is the script that I run 

    import nxt.locator
    
    b = nxt.locator.find_one_brick(name="NXT")
    
    for box in range(10):
        b.message_write(box, 'message test %d' % box)
    for box in range(10):
        local_box, message = b.message_read(box, box, True)
        print local_box, message
        
    print 'Test succeeded!'

this is my .nxt-python file
(I believe that the host should be my bluetooth id)
 
    [Brick]
    name = NXT
    host = 00:16:53:15:BF:19 
    strict = 0
    method = usb=False, bluetooth=True, fantomusb=False

when I run it this is what happens

     python2 messageTest.py 
    Traceback (most recent call last):
      File "messageTest.py", line 3, in <module>
        b = nxt.locator.find_one_brick(name="NXT")
      File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 137, in find_one_brick
        raise BrickNotFoundError
    nxt.locator.BrickNotFoundError

auryan898, 2018-01-04:
```
import nxt.bluesock
b = nxt.bluesock.BlueSock('00:16:53:15:BF:19').connect()
```

find_one_brick never really worked so well for me either, so I just straight up used the bluetooth mac address and directly connected with bluetooth.  find_one_brick was meant to be able to connect to the nxt either with bluetooth or usb, but if you're just using bluetooth this code here can work just as fine.

auryan898, 2018-01-04:
`def find_one_brick(host=None, name=None, silent=False, strict=None, debug=False, method=None, confpath=None)`

I think I see a problem with your code. If you want to use `find_one_brick` with your config file(which seems to be written correctly) you'd have to pass `confpath='./path/to/config/.nxt-python'`

What your code does is look for a brick that has that name `NXT` connected via usb, but there's no usb connection, and so it finds no bricks at all.

pass `debug=True` to see what host and name the `find_one_brick` method is looking to connect to when you run it.

HaginCodes, 2018-01-05:
I like the bluetooth suggestion that you made in your comment with the bluesocket. I will use that its reliable  and works well. I will close this as it working and I will open another one on another topic. Thank you so much for help :)

closed 2018-01-05 by HaginCodes