schodet/nxt-python #122: Motors don't release when program is over.

HaginCodes, 2018-01-05:
Hello I am glad I got this nxt working via bluetooth connection and now I am writing my programs. One problem I am facing is when I run a program and it finishes the motors are still engaged. They aren't loose at the end of a program. When I unplug it I can move the motor freely. But when its plugged in and the program is over it becomes harder to move with my hands to readjust them. How do I fully disengage the motors? Also I am not exactly sure what the two parameters when you do` .turn(100,360)` is the 360 the degrees and the 100 the amount of power? I am not convinced since I can put in negative numbers for the first param which doesn't make sense. Are these the way to do motor delays? or is there a better way to get back to back motor perfomance without getting a `raised BrickException`?

Here is some code I am wondering about.
    
    bucket_degrees = [5,-5,8]
    POWER = 48
    class Tilt_arms():
       
        def spin_bucket(b):
            m_left = Motor(b, PORT_A)
            m_left.turn(bucket_degrees[0], POWER)
            time.sleep(0.5)
            m_right = Motor(b, PORT_A)
            m_right.turn(bucket_degrees[1], POWER)
            time.sleep(0.5)

auryan898, 2018-01-07:
What you need is ´m_right.idle()´. It disables an nxt motor to make it loose again.

HaginCodes, 2018-01-15:
thanks that works :)

HaginCodes, 2018-01-15:
Is there a reason why its so easy to trigger the BlockedException?
    
    def spin_bucket_free_r(b):
        m_right = Motor(b, PORT_B)
 	    m_right.turn(5, 90, True)

nothing is in the way, the more degrees I give the motor it increases in speed a lot and then when it is done it jerks to a halt which is what I want for accuracy but the motor spins so fast.

Eelviny, 2018-02-27:
Hi @HaginCodes, not sure how you mean by the ease of triggering BlockedException, but here's some advice for you:

I've generally found that latency of the connection has a large impact on the accuracy of the motors. Essentially the motor is separated into two parts - the motor itself and a rotation sensor. NXT-Python stops the motor from moving at the exact time it sees that the angle has been reached, but latency means that by the time the NXT has received the stop instruction, the motor has overshot its mark.

In the end this is a hardware limitation, I hope you can work around it! If you can get USB working, latency is much less of an issue.

Eelviny, 2018-02-27:
Closing issue as original problem has been fixed - please open a new ticket for a new issue

closed 2018-02-27 by Eelviny