schodet/nxt-python #123: Unable to run any of the codes or examples

RoboticsNXT, 2018-02-07:
First of all I highly appreciate the efforts of all those involved in making the code.
I am using windows and was trying the nxt-pyhon package with some of the inbuilt examples. But using both Python 2 and 3 I started facing a lot of errors even though I had installed backend packages like Libusb, pyusb in both cases.

For python 2 the error for instance running spin.py package shows-

### Traceback (most recent call last):
  File "C:\Users\apeksha kothari\Downloads\nxt-python-2.2.2\examples\spin.py", line 13, in <module>
    b = nxt.locator.find_one_brick()
  File "C:\Python27\lib\site-packages\nxt\locator.py", line 112, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "C:\Python27\lib\site-packages\nxt\locator.py", line 43, in find_bricks
    for s in socks:
  File "C:\Python27\lib\site-packages\nxt\usbsock.py", line 84, in find_bricks
    for bus in usb.busses():
  File "C:\Python27\lib\site-packages\usb\legacy.py", line 353, in busses
    sorted(core.find(find_all=True), key=lambda d: d.bus),
  File "C:\Python27\lib\site-packages\usb\core.py", line 1263, in find
    raise NoBackendError('No backend available')
NoBackendError: No backend available

For python 3 after correcting statements for python2to3 the error running spin.py package shows-

Traceback (most recent call last):
  File "C:\Users\apeksha kothari\Downloads\nxt-python-2.2.2\examples\spin.py", line 4, in <module>
    import nxt.locator
  File "C:\Users\apeksha kothari\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nxt\__init__.py", line 17, in <module>
    from nxt.sensor import *
  File "C:\Users\apeksha kothari\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nxt\sensor\__init__.py", line 17, in <module>
    from .analog import BaseAnalogSensor
  File "C:\Users\apeksha kothari\AppData\Local\Programs\Python\Python36-32\lib\site-packages\nxt\sensor\analog.py", line 33, in <module>
    class BaseAnalogSensor(Sensor):
NameError: name 'Sensor' is not defined

Please help me out, also let me know in case I want to use Bluetooth what other packages to install as pybluez is not installing on both 2 & 3.

Thank You

auryan898, 2018-02-07:
So first of all, Python 3 will not work whatsoever with nxt-python so don't bother with that. Just stick with python 2.   

That first error that you are getting is because your PyUsb is being finicky. [This thread]
 (https://stackoverflow.com/questions/13773132/pyusb-on-windows-no-backend-available) on StackOverflow might help you. Eelviny also has some build instructions that mention this issue a bit in the [installation wiki](https://github.com/Eelviny/nxt-python/wiki/Installation) for this repo.

I've made a [sample program](https://gist.github.com/auryan898/2a0c2ab7f4a8ba8d4275a3ad0afe2021) for you that you could just copy and paste into your own file.  You'll need the serial number/MAC address of the brick for specificity (or in case you have multiple bricks to connect to).  Here's [Eelviny's instructions](https://github.com/Eelviny/nxt-python/wiki/FAQ) on how to find that MAC address along with how to do other useful things.

auryan898, 2018-02-09:
If you'd like any other help with this or with any more examples, I'd love to help.  I'm still interested in working with this project because it can be a useful tool for myself and some of my peers.  I'll probably start off by making some form of documentation, seeing that there never really was an official documentation put in place.  Maybe I'll set out to maintain a python 2 version of the project too. I don't really know, however. It does seem like a lot of work for just myself. You can find my email on my github profile if you have any questions.

RoboticsNXT, 2018-02-14:
Thank You 
The code worked with Bluetooth but not USB. That's fine.
Secondly, I want to know how to access sound module, basemotor module which expect (object) as input.
Hence what input to give as leaving them blank gives the following error-
Traceback (most recent call last):
  File "C:\Python27\trialmotors2.py", line 12, in <module>
    sound = nxt.Brick(b)
AttributeError: 'module' object has no attribute 'Brick'

auryan898, 2018-02-17:
So that `BaseMotor` _class_ you are talking about has no `__init__()` defined, so you would only be able to make a subclass of it. This is what the `Motor` class does with `BaseMotor ` that appears as the following:
```
class Motor(BaseMotor):
    def __init__(self, brick, port):   #I'll just show these headers for reference
```

`Motor` has an `__init__` function/method so you can make instances of `Motor`.  You can do this in your python scripts:
```
my_motor = nxt.Motor( b , nxt.PORT_A ) #there are nxt.PORT_B and nxt.PORT_C for motors.
```
when reading any headers to use in your own scripts, ignore the `self` and just focus on the `brick` and `port` parameters, or whatever happens to be after the `self`.

If you want some help with code, make a public gist and copy paste it into the gist.  Share the link to it here and I would help with it.

auryan898, 2018-02-17:
BTW, there is no sound module. There is only the:
```
nxt.play_tone_and_wait(  (frequency in Hz)  ,  (millisecond duration)  )
```

You'll have to make your own sound module built around the above function or maybe use the one that I made for my own usage.  Essentially, if you plan on making your sound module, you'll need to run the `play_tone_and_wait` function in a separate `thread` so that you can keep controlling your nxt and separately play music. I use this process in my own modules, but it sometimes slows the nxt down, especially over bluetooth.

RoboticsNXT, 2018-02-19:
On running the following code -

#import nxt-python packages to connect to robot and bluetooth
import nxt, thread, time
import nxt.bluesock # Make sure you remember this!
from nxt.sensor.hitechnic import *
from nxt.brick import Brick
#--Initialize Brick--
b = nxt.bluesock.BlueSock('00:16:53:0F:A2:91').connect() 
#Add the serial number/MAC address of your brick for bluetooth connection
nxt.play_tone_and_wait(440.0, 100) 
# 440Hz is Concert A and 100ms for duration. Basically, make a beep noise to check connection

I am getting the following error - 

Traceback (most recent call last):
  File "C:\Python27\trialmotors2.py", line 10, in <module>
    nxt.play_tone_and_wait(440.0, 100)
AttributeError: 'module' object has no attribute 'play_tone_and_wait'

auryan898, 2018-02-20:
oh darn, it's supposed to say `b.play_tone_and_wait(440.0,100)`
I was just writing that code in a rush that day so I wasn't really checking things!
So sorry!!

auryan898, 2018-02-20:
by the way, you don't have to use `b` for the brick variable.  You can make the bluetooth command look like this `brick_device1 = nxt.bluesock.BlueSock('00:16:53:0F:A2:91').connect()` and then your brick variable is `brick_device1`.  
```
brick_device1 = nxt.bluesock.BlueSock('00:16:53:0F:A2:91').connect()
#You can do:
brick_device1.play_tone_and_wait(440.0, 100)
#OR
left_motor = nxt.Motor(brick_device1,nxt.PORT_C)
```

and also google "GITHUB MARKDOWN CODE BOX".  Basically it would be great if you could be formatting your messages with things like code samples. It just makes it easier.

I just thought I'd add that as a sidenote.

RoboticsNXT, 2018-02-23:
Hey Sir,
Thank You for your help. 
Actually, I am trying to built an autonomous car using neural networks as part of my project. 
So, I wanted to implement PID steering control on the car I built on LEGO. 
I am able to read motor readings using TachoInfo() values.
Could you give me some sample commands to set (steer) the motor using tacho values (if possible with PID control)

auryan898, 2018-02-24:
So I assume that the setup on your lego car is of an orange nxt motor that controls the steering?  Fine nxt motor steering controls using tacho values is not nxt-python's forte.  Because of a combination of inaccurate tacho value, latency in bluetooth connections, and latency in program execution, among other things, you're not going to get very accurate turns using this project.  An [issue](https://github.com/Eelviny/nxt-python/issues/117) #117 was once opened that involved motor turning accuracy for an autonomous nxt robot.  The solution was to use a different framework (not nxt-python) that would run on the nxt itself (no computer attached).  Assuming that your neural network runs in python, we now have a bit of a completely different problem to tackle because iirc, there aren't any python based frameworks that run on the nxt itself...

Issue #117 also contained a solution that said:

**"Create an NXC program, that you start manually on the NXT brick. Create a set of functions that continuously listens to the NXT bluetooth inboxes for a message. When the message and parameters are received, execute the function locally. In this case, a message with the amount of units to move the motor."**

Your project is starting to look a bit more complex, but it is getting that much more interesting as well.  I haven't personally used NXC before, but looking at the docs, it seems like a decent challenge.  For your use case, I would imagine an NXC program that listens for a single message for steering angle, one for power, and I think there are eight more available slots that can be sending sensor data to your computer.  Essentially, nxt-python sends these messages to the NXC program, the NXT executes the commands, NXC sends back sensor information to your (probably python-based) neural network on your computer through nxt-python to process, and then repeat.

Eelviny, 2018-02-27:
@auryan898 summed this up quite nicely - some functions simply can't be replaced by NXT-Python, which is essentially a remote control. For applications requiring very low latency (like very precise motor movements), processing must be done on the brick itself.

Closing as original issue appears solved and is now off-topic

closed 2018-02-27 by Eelviny