schodet/nxt-python #124: Issues with Bluesock bluetooth connection

GolddolphinSKB-Fexcraftian, 2018-02-22:
Good morning y'all, I guess I'm back with another issue regarding types in 3.5.3
This time I'm on a RPi, connected to my brick fine with USB, then with the fast bluesock method with the Brick's MAC adress. This worked fine with connecting, but there appears to be another type issue in the Bluesock class.
My code:
```Python
import nxt.bluesock
from nxt.motor import *
b = nxt.bluesock.BlueSock('00:16:53:18:2A:D1').connect()
m_left = Motor(b, PORT_B)
```
This makes the following error with debug enabled enabled in the Bluesock class.
```Python
Traceback (most recent call last):
  File "/home/pi/Desktop/nxt-stuff/motor test.py", line 4, in <module>
    m_left = Motor(b, PORT_B)
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 227, in __init__
    self._read_state()
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 249, in _read_state
    values = self.brick.get_output_state(self.port)
  File "/usr/local/lib/python3.5/dist-packages/nxt/brick.py", line 27, in poll
    self.sock.send(ogram.bytes())
  File "/usr/local/lib/python3.5/dist-packages/nxt/bluesock.py", line 56, in send
    print(':'.join('%02x' % ord(c) for c in data))
  File "/usr/local/lib/python3.5/dist-packages/nxt/bluesock.py", line 56, in <genexpr>
    print(':'.join('%02x' % ord(c) for c in data))
TypeError: ord() expected string of length 1, but int found```

And this error with debug disabled:

```Python
Traceback (most recent call last):
  File "/home/pi/Desktop/nxt-stuff/motor test.py", line 4, in <module>
    m_left = Motor(b, PORT_B)
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 227, in __init__
    self._read_state()
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 249, in _read_state
    values = self.brick.get_output_state(self.port)
  File "/usr/local/lib/python3.5/dist-packages/nxt/brick.py", line 27, in poll
    self.sock.send(ogram.bytes())
  File "/usr/local/lib/python3.5/dist-packages/nxt/bluesock.py", line 57, in send
    l0 = len(data.encode('utf-8')) & 0xFF
AttributeError: 'bytes' object has no attribute 'encode'```
This is caused by the following code in bluesock.send

```Python
def send(self, data):
        if self.debug:
            print('Send:', end=' ')
            print(':'.join('%02x' % ord(c) for c in data))
        l0 = len(data.encode('utf-8')) & 0xFF
        l1 = (len(data.encode('utf-8')) >> 8) & 0xFF
        d = chr(l0) + chr(l1) + data
        self.sock.send(d)```

The data in this case is bytes, Which I think should instead be a string. I tried converting to a string and then fixing the subsequent error, eventually sending a 3x2 bit hex command to the NXT, but I believe the command had been garbled by my bodged code, as the NXT did nothing until I turned it off.

Any help would be very welcome, I could probably fix myself, I just need some more information as to what the code is supposed to do, manipulating the instruction.

Thanks in advance, Goldsloth

GolddolphinSKB-Fexcraftian, 2018-02-22:
Update to my investigations, Converted the bytes to string.
```Python
    def send(self, data):
        data = data.decode('utf-8')
        print(data)
        if self.debug:
            print('Send:', end=' ')
            print(':'.join('%02x' % ord(c) for c in data))
        l0 = len(data.encode('utf-8')) & 0xFF
        l1 = (len(data.encode('utf-8')) >> 8) & 0xFF
        d = chr(l0) + chr(l1) + data
        self.sock.send(d)
```
Which works now, But puts the error to
```Python
Traceback (most recent call last):
  File "/home/pi/Desktop/nxt-stuff/motor test.py", line 4, in <module>
    m_left = Motor(b, PORT_B)
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 227, in __init__
    self._read_state()
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 249, in _read_state
    values = self.brick.get_output_state(self.port)
  File "/usr/local/lib/python3.5/dist-packages/nxt/brick.py", line 29, in poll
    igram = Telegram(opcode=opcode, pkt=self.sock.recv())
  File "/usr/local/lib/python3.5/dist-packages/nxt/bluesock.py", line 66, in recv
    l0 = ord(data[0])
TypeError: ord() expected string of length 1, but int found
```
Which I shall now try to fix.

GolddolphinSKB-Fexcraftian, 2018-02-22:
I feel like I'm documenting this too much, here's what I think is wrong with this code now:

```Python
    def recv(self):
        data = self.sock.recv(2)
        l0 = ord(data[0])
        l1 = ord(data[1])
        plen = l0 + (l1 << 8)
        data = self.sock.recv(plen)
        if self.debug:
            print('Recv:', end=' ')
            print(':'.join('%02x' % ord(c) for c in data))
        return data
```

Data is now `b'\x19\x00'`

I believe that data should be a string instead.

GolddolphinSKB-Fexcraftian, 2018-02-22:
Upon using the command `m_left.turn(100, 360)` the program crashed with this error:
```Python
Traceback (most recent call last):
  File "/home/pi/Desktop/nxt-stuff/motor test.py", line 5, in <module>
    m_left.turn(100, 360)
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 185, in turn
    self._set_state(state)
  File "/usr/local/lib/python3.5/dist-packages/nxt/motor.py", line 242, in _set_state
    self.brick.set_output_state(*list_state)
  File "/usr/local/lib/python3.5/dist-packages/nxt/brick.py", line 27, in poll
    self.sock.send(ogram.bytes())
  File "/usr/local/lib/python3.5/dist-packages/nxt/bluesock.py", line 54, in send
    data = data.decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte
```

Which makes me think that the byte shouldn't be converted to a string, or that it isn't utf-8... I really need some help to continue my comment spam -.-

auryan898, 2018-02-26:
Have you always been running python 3? How did you manage that? The last working release of nxt-python was python 2...I'd help but I doubt I can if I'm not using the same version

GolddolphinSKB-Fexcraftian, 2018-02-26:
The master branch is the python-2 version run through a convertor to python3, I fixed a few type issues, but USB communication works almost perfectly. Normal bluetooth has issues, also Bluesock is broken as you can see.

GolddolphinSKB-Fexcraftian, 2018-02-27:
Issues solved, Putting in a pull request.

closed 2018-02-27 by Eelviny

dlech, 2018-04-28:
Bluetooth still seems to be broken as of https://github.com/Eelviny/nxt-python/commit/9851bee7417832ec8beea9439632e738d24f578f. I think this issue should be re-opened.

```
$ python3 scripts/nxt_test --verbose
debug = True
Find brick...
Warning: Config file (should be at /home/david/.nxt-python) was not read. Use nxt.locator.make_config() to create a config file.
Host: None Name: None Strict: True
USB: True BT: True
Traceback (most recent call last):
  File "/home/david/work/nxt-python/nxt/locator.py", line 117, in find_one_brick
    info = b.get_device_info()
  File "/home/david/work/nxt-python/nxt/brick.py", line 29, in poll
    igram = Telegram(opcode=opcode, pkt=self.sock.recv())
  File "/home/david/work/nxt-python/nxt/bluesock.py", line 67, in recv
    l0 = ord(data[0])
TypeError: ord() expected string of length 1, but int found
Failed to connect to possible brick
No brick was found.
    Is the brick turned on?
    For more diagnosing use the debug=True argument or
    try the 'nxt_test' script located in /bin or ~/.local/bin
Error while running test:
  File "scripts/nxt_test", line 27, in <module>
    b = nxt.locator.find_one_brick(debug=debug)
  File "/home/david/work/nxt-python/nxt/locator.py", line 157, in find_one_brick
    raise BrickNotFoundError
```

GolddolphinSKB-Fexcraftian, 2018-04-28:
I'll give it another test. Thanks for your work ;)

PR #139 (2018-04-28): fix TypeError when connecting via Bluetooth
This fixes the error:

    TypeError: ord() expected string of length 1, but int found

when connecting via Bluetooth

Tested working on Linux.

Closes #124
commit e8e2ead9 (2018-04-28): fix TypeError when connecting via Bluetooth

This fixes the error:

    TypeError: ord() expected string of length 1, but int found

when connecting via Bluetooth

Tested working on Linux.

Closes #124