schodet/nxt-python #125: Request for information

GolddolphinSKB-Fexcraftian, 2018-02-26:
Good evening, as you could see from #124 ,
I've been messing around with the bluesock method of communicating with the NXT brick.
I would advise you read #124 however disorganised and confused it is...
My enquiry is with the send and receive methods in the bluesock module.
During my testing, I monitored the bytes sent from the host to the brick, which didn't seem to match up with the NXT documentation for direct mode communication.

So, my question is, why is the process of manipulating the bytes needed?
```Python
l0 = len(data.encode('utf-8')) & 0xFF
l1 = (len(data.encode('utf-8')) >> 8) & 0xFF
d = chr(l0) + chr(l1) + data
```
I removed these byte stages, and the sent bytes were exactly what was stated in the NXT documentation, but the brick did nothing, nothing was received.

My assumption is that these bit manipulations are necessary for sending the data via bluetooth to the brick?

I will do some more digging as to what the sock actually does to this data.

Thank you in advance, 
Goldsloth.

GolddolphinSKB-Fexcraftian, 2018-02-26:
Did more digging, found the encoding pattern, which informs me a bit more, still have big questions...
00          01                    02                           03                      04
Length  Message type Message content SUM High Byte SUM Low Byte

These are the definitions of the values:
Length: Ubyte (Length of entire telegram)
Message type: (Defined by message type)
Message content: (Defined for each message)
SUM: Uint16, Negated sum of all previous bytes => Message ID + Message Content + SUM = 0

(Whatever that last line means, it confuses me a lot)

This is pretty confusing stuff, but as far as I can see, this lines up with the bit manipulation seen in the `send` and `receive` methods.
Now I need to find out why there is no response to commands being sent.

GolddolphinSKB-Fexcraftian, 2018-02-27:
Issues solved, Putting in a pull request.

Eelviny, 2018-02-27:
Hey @GolddolphinSKB-Fexcraftian,

Noticed the pull request - basically all documentation for the encoding of bytes has been lost to the void of time, the only base we have left is the code itself. Sorry that I couldn't help you here!

closed 2018-02-27 by Eelviny

tcwan, 2018-03-09:
@GolddophinSKB-Fexcraftian:

Did you manage to get Bluetooth to return responses to commands? 
From my testing in the past using nxt-python 2.2.2, I could not get Bluetooth communications to return responses, but I was using the low level send() and recv() API rather than high level API.

GolddolphinSKB-Fexcraftian, 2018-03-09:
@tcwan I put in a fix for the bluetooth errors, It's in master rn. Just updated the method that encodes data sent to the NXT so it worked properly.

It's working perfectly on RPi now, but still waiting for a fix on lightblue, I might submit an issue if it isn't fixed soon.