schodet/nxt-python #128: Sending messages to NXT

RoboticsNXT, 2018-03-04:
Is there any way wherein I can send messages from python right to NXT's Bluetooth mailbox, while NXT running a program that reads messages from these mailbox can use the data.
Similar to the concept mentioned in this 
https://forums.ni.com/t5/Example-Program-Drafts/Send-Data-from-Computer-to-NXT/ta-p/3501368 
that uses LabVIEW software.

GolddolphinSKB-Fexcraftian, 2018-03-04:
If you read ``NXT Bluetooth Developer Kit`` from https://www.lego.com/en-gb/mindstorms/downloads in the direct commands PDF (Appendix 2) there is shown a command called "Message Write" which enables you to send a message to the NXT's mailboxes, I'm not entirely sure you can send your custom commands to the NXT, but this is possible, in theory if it doesn't already exist. I'll look into it further if you're interested.

Eelviny, 2018-03-05:
@GolddolphinSKB-Fexcraftian NXT-Python will always suffer from the latency issue - for example, turning motors to a precise angle doesn't work well in NXT-Python; by the time the NXT receives the motor stop command, it's usually gone too far. 

In this case, you would instruct a local NXC program to move the motor an exact amount and have the onboard brick process it, creating a hybrid NXC/Python implementation.

Sending messages to NXT inboxes is possible using NXT-Python, but I've forgotten how to and I don't have time to figure it out at the moment...

GolddolphinSKB-Fexcraftian, 2018-03-05:
Given that the NXT-G software seems to be broken on later versions of macOS, I'm not really equipped to figure this out I'm afraid.

dlech, 2018-04-28:
@RoboticsNXT, I believe so. I think the function you are looking for is `nxt.direct.message_write(opcode, inbox, message)`.

baojason, 2019-04-21:
I successfully sent message to NXT mail box with the following code:
from nxt.bluesock import BlueSock
ID = '00:16:53:0B:AE:FD'
sock = BlueSock(ID)
b = sock.connect()
b.message_write(0,"abc")
sock.close()

However, I can never figure out how to read/receive a mailbox message sent by NXT. "message_read(..,true)" always report error like: "nxt.error.DirProtError: Specified mailbox queue is empty", when the nxt loop sending a message out via bluetooth. The following is the method I used:

...
box=0
local, msg = b.message_read(box,box,True) 
...
No matter what kind of value the box is set, the results are the same.

tcwan, 2019-04-24:
> However, I can never figure out how to read/receive a mailbox message sent by NXT. "message_read(..,true)" always report error like: "nxt.error.DirProtError: Specified mailbox queue is empty", when the nxt loop sending a message out via bluetooth. The following is the method I used:
> 

I have never managed to get Bluetooth reads working with nxt-python.
I think if you use USB the message send/receive should work?

baojason, 2019-04-28:
I figure it out. The concept for NXT mailbox communication is based on Master to Slave, which means always polling a slave NXT mailbox for any messages sent by it. The above b.message_read(box, box, True) help to do so, however it should be updated as b.message_read(box+10, box, True). The first parameter is the remote mailbox number according to NXT MESSAGEREAD command found on "LEGO MINDSTORMS NXT Direct Commands" document.

The following code works for me now:
#! /usr/bin/python
ID = '00:16:53:0B:AE:FD'
from nxt.bluesock import BlueSock
import time
import nxt.locator
import nxt.direct

# Create socket to NXT brick
sock = BlueSock(ID)

# On success, socket is non-empty
if sock:

   # Connect to brick
   b = sock.connect()

   box = 0
   i = 0
   while i<100:
       msg = "Msg " + str(i)
       i += 1
       b.message_write(0,msg)
       try:
           time.sleep(0.01) #10ms pause from the docs seems to not be adequate
           local_box, message = b.message_read(box+10, box, True)
           print local_box, message
       except nxt.error.DirProtError, e:
           pass

   # Close socket
   sock.close()

# Failure
else:
   print 'No NXT bricks found'

At the NXT, use NXT-G write a simple slave program that keep reading from bluetooth mailbox 1 and sending it out to bluetooth mailbox 1 with connection 0. The following is the running result of the python script:

0 Msg 0
0 Msg 1
0 Msg 2
0 Msg 3
0 Msg 4
0 Msg 5
0 Msg 6
...
...

tcwan, 2019-04-29:
Hmm. I'm not using the mailboxes, just the usual `send()` and `recv()`, and it has not worked over Bluetooth for me.

baojason, 2019-04-29:
I think one fact about nxt bluetooth communication is that one master NXT can talk to up to 3 slave NXTs, so from master point of view, it doesn't care about conflict. From one slave point of view, it needs to avoid conflict with other slaves' communication. That is why the NXT firmware don't really send slave messages to Bluetooth immediately, instead, messages are put into a queue, and they are send out by Bluetooth only when the master requests to do so.

schodet, 2021-11-05:
This seems to be solved. Closing.

Also if you are interested, I have a page on this subject: https://ni.fr.eu.org/lego/messages/

closed 2021-11-05 by schodet