schodet/nxt-python #150: Issue with NXT communication when trying to "find_one_brick()"

mattschueler, 2018-05-20:
I am trying to write some software that will run on my computer (Windows 10, 64-bit) to allow a group of NXT/EV3s to be controlled by USB controllers. It was relatively easy to get my machine talking to the USB controllers with PyUSB, and now I have been trying to communicate with an NXT for some time, but have not been able to get it to work. My test code for this can be found [here](https://github.com/mattschueler/PyFMS/blob/master/nxt_test.py).

When I run this code (with debug on), all I get is this result-
```
Warning: Config file (should be at C:\Users\Matthew/.nxt-python) was not read. Use nxt.locator.make_config() to create a config file.
Host: None Name: None Strict: True
USB: True BT: True
Failed to connect to possible brick
No brick was found.
    Is the brick turned on?
    For more diagnosing use the debug=True argument or
    try the 'nxt_test' script located in /bin or ~/.local/bin
Traceback (most recent call last):
  File "C:\Users\Matthew\Documents\Python\PyFMS\nxt\locator.py", line 122, in find_one_brick
    info = b.get_device_info()
  File "C:\Users\Matthew\Documents\Python\PyFMS\nxt\brick.py", line 27, in poll
    self.sock.send(ogram.bytes())
  File "C:\Users\Matthew\Documents\Python\PyFMS\nxt\usbsock.py", line 65, in send
    self.device.write(self.blk_out, data, timeout=5000)
  File "C:\WinPython\WinPython-64bit-3.4.4.4\python-3.4.4.amd64\lib\site-packages\usb\core.py", line 948, in write
    self.__get_timeout(timeout)
  File "C:\WinPython\WinPython-64bit-3.4.4.4\python-3.4.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 824, in bulk_write
    timeout)
  File "C:\WinPython\WinPython-64bit-3.4.4.4\python-3.4.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 920, in __write
    _check(retval)
  File "C:\WinPython\WinPython-64bit-3.4.4.4\python-3.4.4.amd64\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 10060] Operation timed out
Bluetooth module unavailable, not searching there
---------------------------------------------------------------------------
BrickNotFoundError                        Traceback (most recent call last)
<ipython-input-1-55ef34b41ad5> in <module>()
     16 else:
     17     # How it is recommended to use nxt-python
---> 18     b = locator.find_one_brick(strict=False, debug=True)

C:\Users\Matthew\Documents\Python\PyFMS\nxt\locator.py in find_one_brick(host, name, silent, strict, debug, method, confpath)
    160     For more diagnosing use the debug=True argument or
    161     try the 'nxt_test' script located in /bin or ~/.local/bin""")
--> 162     raise BrickNotFoundError
    163 
    164 

BrickNotFoundError:
```
I can tell that it is is finding a brick, but that it is timing out when it tries to query the device info from the NXT. I have tried giving it more time but that has not worked. I also had to modify my local copy of the nxt-python library slightly in [this](https://github.com/mattschueler/PyFMS/commit/9aeb7e25708396bf94191e4d991b18bce74d5b43) commit and [this](https://github.com/mattschueler/PyFMS/commit/40aaded00eb588e3f369ad9b7b453de9fedab965) commit to make them work with libusb1.0 (I think).

Any help, no matter how small you might think it is, would be greatly appreciated. If you need any more details as to my project and what i have done, I would be happy to provide them as best I can.

osmarcedron, 2018-05-29:
Apparently, only python2 version is working (more info in other issues), If you ran into problems, use version 2.1.0. Just my two cents.

schodet, 2021-11-16:
The USB code has been updated to use recent PyUSB interface, please reopen if you can still see the problem.

closed 2021-11-16 by schodet