schodet/nxt-python #155: Issue when running example (nxt.error.DirProtError: No active program)

letsgo00, 2019-01-07:
I get this issue when I run a program:

`Traceback (most recent call last):
  File "message_test.py", line 15, in <module>
    b.message_write(box, 'message test %d' % box)
  File "C:\Python27\lib\site-packages\nxt\brick.py", line 31, in poll
    return parse_func(igram)
  File "C:\Python27\lib\site-packages\nxt\direct.py", line 28, in _parse_simple
    tgram.check_status()
  File "C:\Python27\lib\site-packages\nxt\telegram.py", line 114, in check_status
    nxt.error.check_status(self.parse_u8())
  File "C:\Python27\lib\site-packages\nxt\error.py", line 85, in check_status
    raise ex
nxt.error.DirProtError: No active program`

GolddolphinSKB-Fexcraftian, 2019-01-07:
Is there a reason that you're using the python2 version of this project? The python3 version has been more recently maintained, I would advise using that.

This specific error is returned by the brick in response to your command, as you have sent a "message" to the NXT into one of it's inboxes. The NXT has to be running a program in order to deal with these.

(direct.py - `Sends a message to the mailboxes on the NXT brick`)

GolddolphinSKB-Fexcraftian, 2019-01-11:
Which example were you running?

schodet, 2021-11-16:
Python2 version is no longer supported, please try with the master branch and python3.

From what I see from your error message, the NXT is connected and responding.

closed 2021-11-16 by schodet