schodet/nxt-python #157: Problem connecting to two NXT bricks at once via USB

pavel-rybnicek, 2019-05-02:
I'm trying to control two NXT bricks at once. However when I try to connect to the second brick, I get the "resource busy" error. Am I doing anything wrong? Or is that not possible at all?
```
Host: 00:16:53:0E:2F:AE Name: None Strict: True
USB: True BT: True Fantom: False FUSB: False FBT: False
Host: 00:16:53:0A:4D:6E Name: None Strict: True
USB: True BT: True Fantom: False FUSB: False FBT: False
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/nxt/locator.py", line 118, in find_one_brick
    b = s.connect()
  File "/usr/local/lib/python2.7/dist-packages/nxt/usbsock.py", line 45, in connect
    self.handle.setConfiguration(1)
  File "/usr/local/lib/python2.7/dist-packages/usb/legacy.py", line 260, in setConfiguration
    self.dev.set_configuration(configuration)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 794, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy
Failed to connect to possible brick
```
There is no problem with a single brick. The code runs on RaspberryPI (but I guess it doesn't matter).

schodet, 2021-11-16:
Python 2 is no longer supported, but yes I think this is still valid. I manage to connect to two bricks, but I might have been lucky, will try to reproduce.

jrddunbr, 2022-07-17:
I was able to connect 4 bricks at once over USB with no issues:
```
>>> from nxt import locator
>>> bricks = locator.find(find_all=True)
>>> brick_list = []
>>> brick_list.append(next(bricks))
>>> brick_list.append(next(bricks))
>>> brick_list.append(next(bricks))
>>> brick_list.append(next(bricks))
>>> for x in brick_list:
...     x.get_device_info()
... 
('LEDs', '00:16:53:09:52:92', (0, 0, 0, 0), 91912)
('NXT', '00:16:53:09:5F:D8', (0, 0, 0, 0), 91912)
('NXT', '00:16:53:1C:8A:E2', (0, 0, 0, 0), 82516)
('NXT', '00:16:53:16:82:A7', (0, 0, 0, 0), 82260)
```
(Python 3.10.5, Arch Linux)