schodet/nxt-python #158: config file - where is it and how to invoke it for the locator.py

ntnxcrex, 2019-07-16:
Hello All,

I installed all on a raspberry with blouetooth connection.
Examples pythons works randomly and I'm not able to debug.

I have not found the place where to put the configuration file, I used to manually modify the locator.py code to insert my NXT bluethooth address in it.

NXT is paired with raspberry.

where is the located the configuration file I'm reading around?
how to invoke it?
how to set debug mode in the locator.py to understand what's going on?

as I said, sometimes examples pythons like mary.py and test_sensors.py works, sometimes not and i receive the message that raspberry is not able to locate the NXT.

Thanks!

crescenzo

dlech, 2019-07-16:
> where is the located the configuration file I'm reading around?

See https://github.com/Eelviny/nxt-python/wiki/Location-Config (I just edited it to add the file name)

ntnxcrex, 2019-07-22:
Thanks, more clear now.
it worked!
ciao

closed 2021-11-16 by schodet