schodet/nxt-python #161: Cant find brick

Kainkun, 2020-04-04:
Im having trouble getting this to work. When I do brick = nxt.locator.find_one_brick() I get this error

Traceback (most recent call last):
  File "g:\Documents\Python\NXTtest.py", line 4, in <module>
    brick = nxt.locator.find_one_brick()
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\nxt\locator.py", line 112, in find_one_brick
    for s in find_bricks(host, name, silent, method):
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\nxt\locator.py", line 43, in find_bricks
    for s in socks:
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\nxt\usbsock.py", line 84, in find_bricks
    for bus in usb.busses():
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\legacy.py", line 352, in <genexpr>
    return (Bus(g) for k, g in groupby(
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\legacy.py", line 347, in __init__
    self.devices = [Device(d) for d in devices]
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\legacy.py", line 333, in __init__
    self.configurations = [Configuration(c) for c in dev]
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\core.py", line 1094, in __iter__
    yield Configuration(self, i)
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\core.py", line 586, in __init__
    configuration
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\backend\libusb1.py", line 757, in get_configuration_descriptor  
    config, byref(cfg)))
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\usb\backend\libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found

auryan898, 2020-04-05:
find_one_brick is sometimes difficult to work with, so if you are using exclusively just bluetooth or just usb, I suggest using the specific methods to do that. It seems that your code is attempting to connect via USB, but could not detect it. Is it plugged in? Will you do so? I assume LibUsb is installed, but it is just having trouble connecting.

Here is a past issue troubleshooting bluetooth connections, if that is what you are attempting: #121

Kainkun, 2020-04-05:
Thanks, I thought that  find_one_brick is the only way to access the brick.

I tried this
b = nxt.bluesock.BlueSock('00:16:53:07:6a:0b').connect()
and got this.

Traceback (most recent call last):
  File "g:\Documents\Python\NXTtest.py", line 7, in <module>
    b = nxt.bluesock.BlueSock('00:16:53:07:6a:0b').connect()
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\nxt\bluesock.py", line
39, in connect
    sock = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
  File "C:\Users\Kainkun\Anaconda2\lib\site-packages\nxt\lightblueglue.py",
line 34, in __init__
    _sock = lightblue.socket(proto)
AttributeError: 'module' object has no attribute 'socket'

Do you know whats going on? As for USB I'm trying what this is doing
https://stackoverflow.com/questions/51603286/python-nxt-mindstorms-connection-issues

but I'm not sure how to set the config file

If there are different methods to connect please let me know.
Thank you so much

schodet, 2021-11-16:
It seems that you do not have PyBluez installed so it falls back to lightblue.

Can you try with PyBluez installed?

Please reopen if you still have the problem with python3, the master branch, and PyBluez.

closed 2021-11-16 by schodet