schodet/nxt-python #163: Python 3 port

skippa, 2020-08-11:
The readme mentions v3.xx and that it won't be compatible with old v2 scripts - has anyone converted this to Python 3 yet? Getting pip to install the requirements into correct Python 2 directories seems to be a pain on a system with Python 3 and Python 2 installed.

I'm close to setting up a docker image to save hassles... I'm running Arch Linux BTW ;)

GolddolphinSKB-Fexcraftian, 2020-08-12:
I'm fairly sure the master branch works on Python3 , there's a seperate branch taken before the change for py2.

skippa, 2020-08-12:
Ah yes, I've since realized that install.sh installs python 3 version (python setup.py install does python 2 version), however some examples bread to be converted using 2to3

closed 2020-08-14 by skippa