schodet/nxt-python #170: Tester on Windows wanted

schodet, 2021-11-16:
Hi!

I am looking for testers with some python background on Windows system.

Please contact me if you can help.

dbrown3562, 2022-04-17:
Good evening! If this issue is still open, I can help. I'm a Window users with a fair amount of Python experience.

poggersv2, 2023-04-26:
Hey, I a windows 11 user and have moderate python experience. I think I can help...

schodet, 2023-05-04:
Thanks for your help!

My comment above from last year is still valid :)

What would be nice would be to do a test matrix, and a guide to connect nxt-python over bluetooth. Can you help with that?

poggersv2, 2023-05-08:
Hey, unfortunately my nxt just broke a few days ago due to me dropping it to the ground multiple times. Shipping to where I live takes a month or two, so I don't think I'll be able to help anymore.

schodet, 2023-05-08:
OK, thanks for your answer.

LakshmiGayathri19, 2023-11-10:
Hello @schodet ,
Do you still need help with the above issue ? If yes, I am willing to contribute. Please do let me know.
FYI: I am a windows 11 user and i am proficient in python.

Thanks in advance!

SarojArchana, 2023-11-16:
Hi @schodet schodet, is the issue still open, I would be happy to help you out.

schodet, 2023-11-16:
Hello @LakshmiGayathri19 and @SarojArchana, sorry for the late response.

Yes, the issue is still open, I would be glad if you can test what works and what does not work on Windows and improve the documentation in the guide.

Well, and if something does not work, help to fix it :)

Thanks.

LakshmiGayathri19, 2023-11-16:
Thank you for the response. Will start the testing and update you regarding the same periodically.

schodet, 2024-01-02:
Hi, how is it going? Did you find stop points?

Thanks

schodet, 2026-01-07:
If you do not have an NXT, this will be hard…