schodet/nxt-python #171: Tester on mac OS wanted

schodet, 2021-11-17:
Hi!

I am looking for testers with some python background on mac OS system.

Please contact me if you can help.

JoseSantos7, 2021-12-15:
I am available for this one, just have to overcome the setup of installing and putting it to work

hope I can help

michthom, 2021-12-22:
I'd love to help, but sadly I found this repo just before I found that my NXT brick has the dead display issue, so currently I'm unable to do much until I get a replacement for the screen connection... best of luck however!

schodet, 2021-12-27:
You can use NXT-Python even without a screen :)

michthom, 2021-12-28:
Thanks Nicolas, but I’m struggling to get the Bluetooth pairing to complete as I can’t see the NXT side of the conversation. I have ordered some conductive adhesive and hope to have my screen working with a new ribbon cable in a couple of weeks when that arrives.

rhart-appriss, 2022-06-11:
@schodet I have a working NXT brick on macOS. I had trouble installing pybluez from pip, but was able to install it from the latest on github. Let me know if I can help update the documentation or provide any testing

rrhartjr, 2022-06-11:
@schodet I have a working NXT brick on macOS. I was not able to get pybluez installed from pip, but was able to install it using latest on github. Let me know if you need help with documentation or testing

schodet, 2022-06-13:
@rrhartjr Thanks for your help!

Which macOS version are you using?

Did you manage to make USB work?

About documentation, could you add a section describing how you installed pybluez?

Thank you :)

rrhartjr, 2022-06-18:
> @rrhartjr Thanks for your help!
> 
> Which macOS version are you using?

Works on Catalina and Big Sur

> 
> Did you manage to make USB work?
Works with no issues per documentation
> 
> About documentation, could you add a section describing how you installed pybluez?
PR opened
> 
> Thank you :)

tcwan, 2022-06-21:
@schodet I've not been frequenting github so I missed this.
Anyway, I've been running python3+nxt-python 3.0.0 on macOS for over 6 months now.

Works fine with libusb + pyusb (tested with python 3.9 from MacPorts, didn't try with later versions). Got it working on both Intel Macs running 10.14 (Mojave), and Intel + M1 Macs running 12.x (Monterey).

Didn't try bluetooth support, due to the fact that the pybluez package is not available on MacPorts and I've not managed to get bluetooth (lightblue package) to work with my nxos-armdebug stuff in the past. Wasn't sure if it was worth the effort to debug it.

tcwan, 2022-06-22:
> I may be useful to have a “tested matrix”, what do you think?

Sure, though it will probably be a bit sparse, I only have access to certain versions of macOS.
Mojave was available mainly because it's the last 32-bit compatible version of macOS, and running on rather old iMacs (~2011 vintage).

schodet, 2022-06-26:
So for the moment, it looks like this:

| macOS | Python | USB | Bluetooth |
|------|-----|-----|-----|
| 10.14 (Mojave) | 3.9 (MacPorts) | :heavy_check_mark: | ? |
| 10.15 (Catalina) | | :heavy_check_mark: | :heavy_check_mark: * | 
| 11 (Big Sur) | | :heavy_check_mark: | :heavy_check_mark: * |
| 12 (Monterey) | 3.9 (MacPorts) | :heavy_check_mark: | ? |

*: pybluez build from master

schodet, 2022-06-26:
@tcwan How is the Portfile used? Is it something to include in nxt-python, in a contrib directory for example?

tcwan, 2022-08-09:
> @tcwan How is the Portfile used? Is it something to include in nxt-python, in a contrib directory for example?

It's used for [MacPorts](https://guide.macports.org/chunked/development.local-repositories.html). It should be in a contrib directory.