schodet/nxt-python #172: Assistance required during installation process

jackwalsh5801, 2021-11-28:
hello,
I am a new beginner when it comes to python, but very experienced with legos official software. i figured after spending years programming using word blocks that i would make the next step towards learning python. my question is simple, how do i install NXT - python? 
i have honestly spent hours trying to understand how to even begin the installation process, and i still cannot understand how to install it after ive installed numerous versions of python.
could you please assist me in translating the installation process into a words that are understandable to an extremely new beginner?
Thank you

schodet, 2021-11-28:
First, keep in mind that NXT-Python is running on your computer, not on the brick, which makes a real difference. You can use NXC to program the NXT on the brick with a text based language.

For NXT-Python, first make sure that you have a working python3 installation, then clone this repository:

    git clone https://github.com/schodet/nxt-python.git

And install the module:

    cd nxt-python
    python3 setup.py install

For a simpler method, you will have to wait a little bit that I publish the first python 3 release (I expect to do it in December), then installation instructions will be simpler.

JoseSantos7, 2021-12-12:
hello

I am also new to this - specifically python and lego nxt . I recently have been handled a lego nxt Mindstorms set

I expect this could be an enriching source; I don't know how to put this to work in python + Mac

sorry for the ignorance, what do you mean by:

And install the module:

cd nxt-python
python3 setup.py install

thank you for your help

schodet, 2021-12-12:
These are commands to run from the command line, change to the directory you downloaded nxt-python, then run the setup.py script with the install parameter.

You need to learn how to use Python first if you want to use NXT-Python.

auryan898, 2021-12-12:
@JoseSantos7 
So first go to the Python website and install "Python 3.8, 3.9, or 3.10" for Mac
Now go and install "Git VCS"

schodet, 2021-12-13:
Keep in mind that I plan to do an alpha release in December, there will be some interface changes.

JoseSantos7, 2021-12-13:
thank you both

it installed everything except the error below. pybluez is to make possible the connection via bluetooth?

do you know why?

message conveyed:

Download error on https://pypi.org/simple/pybluez/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997) -- Some packages may not be found!
Couldn't find index page for 'pybluez' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
Download error on https://pypi.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997) -- Some packages may not be found!
No local packages or working download links found for pybluez>=0.23
error: Could not find suitable distribution for Requirement.parse('pybluez>=0.23')

schodet, 2021-12-13:
Yes, pybluez is only for Bluetooth.

There seems to be something wrong with your installation, is your OS up to date? Are you using a proxy which filter HTTPS?

schodet, 2021-12-13:
This means that something is broken on your computer or network:

```
Download error on https://pypi.org/simple/pybluez/: [SSL: CERTIFICATE_VERIFY_FAILED]
```

You could try again, and include a full log of what command you typed and what
was the result, but this has nothing to do with NXT-Python.

By the way, I made the alpha release, so you can try it like this:

```
python3 -m pip install --upgrade --pre nxt-python
```

Here is what I get on my computer:

```
nicolas@boktor:~/work/lego/nxt-python% python3 -m pip install --upgrade --pre nxt-python
Collecting nxt-python
  Downloading nxt_python-3.0.0a1-py3-none-any.whl (74 kB)
     |████████████████████████████████| 74 kB 1.9 MB/s
Collecting pybluez>=0.23
  Using cached PyBluez-0.23.tar.gz (97 kB)
Collecting pyusb>=1.2.1
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Using legacy 'setup.py install' for pybluez, since package 'wheel' is not installed.
Installing collected packages: pyusb, pybluez, nxt-python
    Running setup.py install for pybluez ... done
Successfully installed nxt-python-3.0.0a1 pybluez-0.23 pyusb-1.2.1
nicolas@boktor:~/work/lego/nxt-python%
```

Please use triple back-quotes to insert the log like I just did.

JoseSantos7, 2021-12-13:
it is somewhat different from yours...

```Collecting pre
  Downloading pre-1.3.0.tar.gz (704 bytes)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: nxt-python in /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages (2.2.2)
Using legacy 'setup.py install' for pre, since package 'wheel' is not installed.
Installing collected packages: pre
    Running setup.py install for pre ... done
Successfully installed pre-1.3.0```

schodet, 2021-12-13:
You installed the package `pre`… Just copy/paste the command line.

JoseSantos7, 2021-12-14:
thank you for your help

I am doing something wrong and I don't get what it is....

'''python3 -m pip install --upgrade --pre nxt-python
Requirement already satisfied: nxt-python in /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages (2.2.2)
Collecting nxt-python
  Using cached nxt_python-3.0.0a1-py3-none-any.whl (74 kB)
Collecting pyusb>=1.2.1
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Collecting pybluez>=0.23
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Library/Frameworks/Python.framework/Versions/3.10/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/_z/m60znzld66189n8_4_83xr8m0000gn/T/pip-install-wu8lqimg/pybluez_f98e4e47e4c14b73b87b61b2de157283/setup.py'"'"'; __file__='"'"'/private/var/folders/_z/m60znzld66189n8_4_83xr8m0000gn/T/pip-install-wu8lqimg/pybluez_f98e4e47e4c14b73b87b61b2de157283/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/_z/m60znzld66189n8_4_83xr8m0000gn/T/pip-pip-egg-info-xu6ffga2
       cwd: /private/var/folders/_z/m60znzld66189n8_4_83xr8m0000gn/T/pip-install-wu8lqimg/pybluez_f98e4e47e4c14b73b87b61b2de157283/
  Complete output (1 lines):
  error in PyBluez setup command: use_2to3 is invalid.
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/08/9f/e9d93b266d2d1ea988780a52a696073ba0a65df65a532165fdf6ff90d0ed/PyBluez-0.23.tar.gz#sha256=c8f04d2e78951eaa9de486b4d49381704e8943d0a6e6e58f55fcd7b8582e90de (from https://pypi.org/simple/pybluez/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.'''

schodet, 2021-12-14:
I do not think it’s you, there is several issues open on PyBluez for python 3.10.

For example: https://github.com/pybluez/pybluez/issues/424

Could you try with python 3.7?

----

Also, I am not sure whether it installed nxt-python from a previously built cached package or if it comes from PyPI, as the version in the git repo was also 3.0.0a1. You can try with `--no-cache-dir` and `--force-reinstall` if this is a problem:

```
python3 -m pip install --upgrade --no-cache-dir --force-reinstall --pre nxt-python
```

----

Are you able to use the USB connection? If you run `nxt_test`, it should detect the NXT brick and play a sound.

I am writing a NXT-Python tutorial for the next alpha release.

JoseSantos7, 2021-12-15:
hello

honestly, thank you for your dedication and help

yes, usb is good

sorry for the ignorance, I went to directory nxt-python  and I can't find any program 'nxt-test.py' to run

I have others like motor.py, brick.py, etc

thank you

schodet, 2021-12-15:
If nxt-python is installed correctly, `nxt_test` should be in your `$PATH`, this means that you can run in from anywhere just by typing `nxt_test` on the command prompt.

If that does not work, you can find it in scripts:

```
cd "where you have cloned nxt-python"/scripts
./nxt_test
```

This one: https://github.com/schodet/nxt-python/blob/master/scripts/nxt_test

JoseSantos7, 2021-12-15:
ok

the brick is turned on and with usb cable connected

after ```nxt-test```this is what happens

```Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 573, in _build_master
    ws.require(__requires__)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 891, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 782, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (nxt-python 2.2.2 (/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages), Requirement.parse('nxt-python==3.0.0a1'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/nxt_test", line 4, in <module>
    __import__('pkg_resources').run_script('nxt-python==3.0.0a1', 'nxt_test')
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3267, in <module>
    def _initialize_master_working_set():
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3241, in _call_aside
    f(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3279, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 575, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 588, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pybluez>=0.23' distribution was not found and is required by nxt-python```

last lines is misleading, seems it is still looking for pybluez

schodet, 2021-12-15:
This looks strange too:

```
pkg_resources.VersionConflict: (nxt-python 2.2.2 (/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages), Requirement.parse('nxt-python==3.0.0a1'))
```

It looks like there is a 2.2.2 version installed somewhere.

----

I made a 3.0.0a2 version which drops the pybluez requirement, can you try to install it?

Once this works I may have another solution for Bluetooth.

JoseSantos7, 2021-12-17:
hi there

I am struggling a little bit, so I will remove everything ok and reinstall again

hope it works

via usb, there is nothing to, running a script without any further action will work in the brick?

thank you

schodet, 2021-12-17:
Yes, it should work without any further action.

JoseSantos7, 2021-12-17:
hi there

it seems not be looking anymore for pybluez but everything else seems the same, after I got rid of and reinstalled eveyrhing again (python 3.10 from their site and nxt package from here)

don't know why the conflict version: ```pkg_resources.VersionConflict: (nxt-python 2.2.2 (/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages), Requirement.parse('nxt-python==3.0.0a2'))```

the entire reaction is this:

```Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 573, in _build_master
    ws.require(__requires__)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 891, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 782, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (nxt-python 2.2.2 (/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages), Requirement.parse('nxt-python==3.0.0a2'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/nxt_test", line 4, in <module>
    __import__('pkg_resources').run_script('nxt-python==3.0.0a2', 'nxt_test')
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3267, in <module>
    def _initialize_master_working_set():
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3241, in _call_aside
    f(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3279, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 575, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 588, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pkg_resources/__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pyusb>=1.2.1' distribution was not found and is required by nxt-python
```

schodet, 2021-12-17:
It seems that there is a 2.2.2 version in `/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages`. 

Can you try to find help from someone who knows mac OS? I suppose that this directory was not emptied when you got rid of everything.

JoseSantos7, 2021-12-18:
yes, thank you

@auryan898 are you proficient with this?

can you help me overcome this issue? the version conflict and the rest...

thank you

Svein-Tore, 2021-12-21:
Hi!
As many other here I have problems with the installation of the nxt-python.  After trying a lot of things I thought it was a good idea to reinstall, and ran the folllwoing commad:

python3 -m pip install --upgrade --no-cache-dir --force-reinstall --pre nxt-python 
Result:
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting nxt-python
  Downloading https://files.pythonhosted.org/packages/58/7c/51804d620621a661a1c5c39c15245dbfe3d21dc7faf0539c97ac81fa0cdc/nxt_python-3.0.0a1-py3-none-any.whl (74kB)
    100% |████████████████████████████████| 81kB 4.2MB/s 
Collecting pyusb>=1.2.1 (from nxt-python)
  Downloading https://files.pythonhosted.org/packages/15/a8/4982498b2ab44d1fcd5c49f07ea3795eab01601dc143b009d333fcace3b9/pyusb-1.2.1-py3-none-any.whl (58kB)
    100% |████████████████████████████████| 61kB 5.1MB/s 
Collecting pybluez>=0.23; platform_system == "Linux" (from nxt-python)
  Downloading https://www.piwheels.org/simple/pybluez/PyBluez-0.23-cp37-cp37m-linux_armv7l.whl (132kB)
    100% |████████████████████████████████| 133kB 1.3MB/s 
Installing collected packages: pyusb, pybluez, nxt-python
  Found existing installation: pyusb 1.2.1
    Uninstalling pyusb-1.2.1:
      Successfully uninstalled pyusb-1.2.1
  Found existing installation: PyBluez 0.23
    Uninstalling PyBluez-0.23:
      Successfully uninstalled PyBluez-0.23
  Found existing installation: nxt-python 3.0.0a1
    Uninstalling nxt-python-3.0.0a1:
      Successfully uninstalled nxt-python-3.0.0a1
Successfully installed nxt-python-3.0.0a1 pybluez-0.23 pyusb-1.2.1

Going through this inof it seems to me  I have successfull installation. But then by executing nxt_test I got the following:
pi@raspberrypi:~ $ nxt_test
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 7, in <module>
    import nxt.locator
  File "/home/pi/.local/bin/nxt.py", line 3, in <module>
    import nxt.locator
ModuleNotFoundError: No module named 'nxt.locator'; 'nxt' is not a package
I go to folder /hom/pi/.local/bin andtry again:

pi@raspberrypi:~ $ nxt_test
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 7, in <module>
    import nxt.locator
  File "/home/pi/.local/bin/nxt.py", line 3, in <module>
    import nxt.locator
ModuleNotFoundError: No module named 'nxt.locator'; 'nxt' is not a package

I try with python command:
pi@raspberrypi:~/.local/bin $ python nxt_test
  File "nxt_test", line 20
    print('Find brick...', flush=True)
                                ^
SyntaxError: invalid syntax
pi@raspberrypi:~/.local/bin $ 

And with python3 command:
i@raspberrypi:~/.local/bin $ python3 nxt_test
Traceback (most recent call last):
  File "nxt_test", line 7, in <module>
    import nxt.locator
  File "/home/pi/.local/bin/nxt.py", line 3, in <module>
    import nxt.locator
ModuleNotFoundError: No module named 'nxt.locator'; 'nxt' is not a package
pi@raspberrypi:~/.local/bin $ 

I have to mention that  I have python 3.7 and I'm running a RPi 4.   I have only tried with bluetooth, not with USB connection.

schodet, 2021-12-21:
> pi@raspberrypi:~/.local/bin $ nxt_test Traceback (most recent call last): File "/home/pi/.local/bin/nxt_test", line 7, in import nxt.locator ImportError: bad magic number in 'nxt': b'\x03\xf3\r\n'
> 
> Any suggetions?

You seems to have a nxt or nxt.py file in .local/bin, where does it come from? Not from nxt-python.

Please use triple backquotes when pasting in github. Also, you are hijacking a issue for an unrelated problem, please use the mailing list.

schodet, 2021-12-21:
@JoseSantos7 look at your `/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages` directory, there must be an old nxt-python version here, please remove it and retry.

schodet, 2022-02-13:
Closing, please recontact me if you need more help.

closed 2022-02-13 by schodet