schodet/nxt-python #174: Problems with installing nxt-python

Svein-Tore, 2021-12-22:
- OS: linux
- Python version:   v3.7
- NXT-Python version: unsure, installed with comand python3 -m pip install --upgrade --pre nxt-python  21 dec 2021
- Connection method: USB? Bluetooth? bluetooth

After installing nxt-python it said it was successfull.   When I run th nxt_test i get the following message:

pi@raspberrypi:~/.local/bin $ nxt_test
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 10, in <module>
    p.add_argument("--sound", action=argparse.BooleanOptionalAction, default=True,
AttributeError: module 'argparse' has no attribute 'BooleanOptionalAction'

Any suggestions?

schodet, 2021-12-22:
`BooleanOptionalAction` does not exists in python 3.7, I will make a release in a few day to change this.

schodet, 2021-12-27:
I made a 3.0.0a3 release with 3.7 support. Are you still able to test it?

Svein-Tore, 2021-12-28:
Hi

Nicolas! As I have mentioned in issue#174, my rpi now have pyghon 3.9.
So thats nit so easy to do.  Lets hope I can run nxt-python with python
3.9👍

Svein Tore

schodet, 2021-12-28:
OK, I close this issue then.

closed 2021-12-28 by schodet