schodet/nxt-python #175: Error when  connecting with bluetooth

Svein-Tore, 2021-12-22:
I have created a simple program found in the nxt-python docs:

import nxt.locator
from nxt.motor import *
#00:16:53:11:2D:2C my NXTs bluetooth MAC adress
with nxt.locator.find(host="00:16:53:11:2D:2C") as b:
    print(b.get_device_info()[0:2])
I ran the program with resuilt (NXT paired with rawspberry PI):

pi@raspberrypi:~/.local/bin $ python3 test.py
failed to connect to device Bluetooth (00:16:53:11:2D:2C)
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    with nxt.locator.find(host="00:16:53:11:2D:2C") as b:
  File "/home/pi/.local/lib/python3.7/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

So  I tried without pairing, and I got a differnt result:

pi@raspberrypi:~/.local/bin $ python3 test.py
failed to connect to device Bluetooth (00:16:53:11:2D:2C)
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    with nxt.locator.find(host="00:16:53:11:2D:2C") as b:
  File "/home/pi/.local/lib/python3.7/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

This time ig get a sound from the NXT, and on the NXT screen it stands:

passkey:
1234_

I press the orange button on the NXT and the error message comes on the Raspberry PI.  

What might be wrong?  Should  I try with USB cable?  Then I guess I must modify the program?

Svein-Tore, 2021-12-22:
So I have tested with debugging, then I get some more info:

pi@raspberrypi:~/.local/bin $ python3 test.py
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/pi/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:16:53:11:2D:2C)
WARNING:nxt.backend.bluetooth:failed to connect to device Bluetooth (00:16:53:11:2D:2C)
DEBUG:nxt.backend.bluetooth:error from connect
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (13, 'Permission denied')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/nxt/backend/bluetooth.py", line 127, in find
    brick = sock.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/nxt/backend/bluetooth.py", line 52, in connect
    sock.connect((self._host, PORT))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 13] Permission denied
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    with nxt.locator.find(host="00:16:53:11:2D:2C") as b:
  File "/home/pi/.local/lib/python3.7/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

So we can see: bluetooth.btcommon.BluetoothError: [Errno 13] Permission denied

Is this a known issue?

schodet, 2021-12-22:
> What might be wrong? Should I try with USB cable? Then I guess I must modify the program?

No need to change the program, it should find the brick and check its address.

> So we can see: bluetooth.btcommon.BluetoothError: [Errno 13] Permission denied

I have never seen this error with bluetooth, what  system is installed on your raspberrypi? I may try to reproduce it here

Svein-Tore, 2021-12-22:
I think my system is Debian, not quite sure. Do you need some more info from me to reproduce it?

I used the command python3 -m pip install --upgrade --no-cache-dir --force-reinstall --pre nxt-python

Found that as an advise in issue  #174.

Svein-Tore, 2021-12-23:
Now I have checked it out, the system is raspbian, a version of debian  for raspberry pi.   I'm doing an upgrade of the pi now, haven't done that for over a year. It might help. I have scanned through the docs for nxt-pyhton and I'm really looking forward to get started.  I have worked with lego mindstorms since 1999.  I have only used LabVIEW, but this version looks really interesting.

Svein-Tore, 2021-12-23:
Shouldn't have done that. Now the PI won't start.........

schodet, 2021-12-23:
> Shouldn't have done that. Now the PI won't start.........

I hope you did not lost any data. Installing a recent version should be
easy enough if you have a sd card writer on your computer. You will get
python 3.9.

Svein-Tore, 2021-12-23:
I haven't lost any data. But I have to install opencv also and nxt-python.
My experince is that this is time consuming... Hope I cn fix it in time. My
students should start this project 3. Januar hopefully.......

schodet, 2021-12-27:
Did you managed to make it work on a recent install?

Svein-Tore, 2021-12-28:
Hi Nicolas!

Thanks for your support!

The pi is up and running and I managed to install opencv.  I also managed
to install nxt-python and it said it was successfully.  But I didn't get
the nxt_test to work. But I have not tested it good enough to conclude.
Btw the rpi now has python 3.9.  At the moment I'm on a short vacation and
I don't have any rpi available.  But on thursday I will look into it. The
error was it couldn't find any brick.

Svein-Tore, 2021-12-30:
So I have tried to connect with the nxt_test, but I got the follwoing result (python 3.9 is installed):

pi@raspberrypi:~/.local/bin $ python3 nxt_test
Find brick...
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 21, in <module>
    with nxt.locator.find_with_options(options) as b:
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 289, in find_with_options
    return find(find_all=find_all, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

schodet, 2021-12-31:
Do you have more info? How is the brick connected? Did you try to increase log level?

Svein-Tore, 2022-01-01:
Hi!

Tried --log:DEBUG, and then got a bit more info:
pi@raspberrypi:~/.local/bin $ python3 nxt_test --log-level=DEBUG
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/pi/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.backend.bluetooth:no bluetooth module
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 21, in <module>
    with nxt.locator.find_with_options(options) as b:
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 289, in find_with_options
    return find(find_all=find_all, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

How is the brick connected?  I'm trying to connect with bluetooth.  But I'm not sure, do I have to pair the NXT and the RPi before running the script?  who is gonna to be master/slave?

schodet, 2022-01-01:
`INFO:nxt.backend.bluetooth:no bluetooth module`

On debian, the simplest solution is : `sudo apt install python3-bluez`

You have to pair the NXT, it will be the device and you computer will be the initiator. Can you take screenshots of the pairing process?

Svein-Tore, 2022-01-01:
In bluetoothctl  I get:

Device 00:16:53:11:2D:2C RSSI: -36

What does RSSI:-36 mean?

When trying to pair with bluetoothctl I get this:

bluetooth]# pair 00:16:53:11:2D:2C
Attempting to pair with 00:16:53:11:2D:2C
Failed to pair: org.bluez.Error.AlreadyExists

Does it mean it is already paired?

I ran the following comamnd:  sudo apt install python3-bluez

When runnig nxt_test I got:
pi@raspberrypi:~/.local/bin $ python3 nxt_test
Find brick...
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 21, in <module>
    with nxt.locator.find_with_options(options) as b:
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 289, in find_with_options
    return find(find_all=find_all, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

On the other hand, I have created a very simple program:

import nxt.locator
from nxt.motor import *

#00:16:53:11:2D:2C is my NXTs bluetooth MAC adress

with nxt.locator.find(host="00:16:53:11:2D:2C") as b:
    print(b.get_device_info()[0:2])

When I run this program  I get:

pi@raspberrypi:~/.local/bin $ python3 connection.py
('Velocita', '00:16:53:11:2D:2C')

Which means this script runs perfect, am I right? strange....

Svein-Tore, 2022-01-01:
Hi Nicolas!

Things are getting better! This program works now:

import nxt.locator
from nxt.motor import *
#00:16:53:11:2D:2C my NXTs bluetooth MAC adress
def spin_around(b):
    m_left = Motor(b, Port.B)
    m_left.turn(100, 360)
    m_right = Motor(b, Port.C)
    m_right.turn(-100, 360)
with nxt.locator.find(host="00:16:53:11:2D:2C") as b:
    spin_around(b)

I can't be sure, but maybe there is something wrong with the nxt_test?

schodet, 2022-01-01:
Good news it works for you :)

> Does it mean it is already paired?

Yes, I think it is.

Can you run this:

- Turn the NXT on,
- `bluetoothctl`, then `devices`, then `exit`
- `echo $PATH`
- `nxt_test --log-level=debug`
- `python3 nxt_test --log-level=debug`
- `python3 nxt_test --log-level=debug --host=00:16:53:11:2D:2C`

And copy/paste the full terminal log please?

Thanks.

Svein-Tore, 2022-01-01:
Here is the full terminal log:

pi@raspberrypi:~/.local/bin $ nxt_test --log-level=debug
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/pi/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 21, in <module>
    with nxt.locator.find_with_options(options) as b:
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 289, in find_with_options
    return find(find_all=find_all, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found
pi@raspberrypi:~/.local/bin $ python3 nxt_test --log-level=debug
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/pi/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 21, in <module>
    with nxt.locator.find_with_options(options) as b:
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 289, in find_with_options
    return find(find_all=find_all, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
pi@raspberrypi:~/.local/bin $ python3 nxt_test --log-level=debug --host=00:16:53:11:2D:2D
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/pi/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:16:53:11:2D:2D)
WARNING:nxt.backend.bluetooth:failed to connect to device Bluetooth (00:16:53:11:2D:2D)
DEBUG:nxt.backend.bluetooth:error from connect
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (112, 'Host is down')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/backend/bluetooth.py", line 127, in find
    brick = sock.connect()
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/backend/bluetooth.py", line 52, in connect
    sock.connect((self._host, PORT))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 112] Host is down
Traceback (most recent call last):
  File "/home/pi/.local/bin/nxt_test", line 21, in <module>
    with nxt.locator.find_with_options(options) as b:
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 289, in find_with_options
    return find(find_all=find_all, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found
pi@raspberrypi:~/.local/bin $ 

Things works great for me. Though I have problem with getting two motors to run a the same time.  May I ask for it here, or open an new issue (question)?

schodet, 2022-01-01:
There is a typo in your command, can you copy/paste?

For the two motors, you need to use threads, or `SynchronizedMotors`, but, this code does not seems really mature.

Another option is to have a NXC program on the NXT and you exchange messages with it.

But yes, this is another subject.

schodet, 2022-01-01:
Also, what about the part about `bluetoothctl`?

schodet, 2022-01-02:
It seems there is someone with the same problem (discovery not working, but direct connection working) on raspberry pi, what is your raspberry pi model?

Svein-Tore, 2022-01-02:
Hi Nicolas!

I use RPI version  4.  Sorry for forgetting the bluetoothctl:

i@raspberrypi:~ $ bluetoothctl
Agent registered
[bluetooth]# devices
Device 00:16:53:11:2D:2C Velocitatem
[bluetooth]# info 00:16:53:11:2D:2C
Device 00:16:53:11:2D:2C (public)
	Name: Velocitatem
	Alias: Velocitatem
	Class: 0x00000804
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
[bluetooth]# exit

for me everything seems to be OK
I also lost some more parts:
pi@raspberrypi:~ $ echo $PATH
/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

I'm not sure what you meant with:
 here is a typo in your command, can you copy/paste?

When it comesto the tweo motor problem, my plan is as follow:

create a sync motor program in labvirw, then copy the program to the NXT as an .rxe.  

According to the NXT-python API:

Brick.start_program(name)

Start a program on the brick.

Parameters
name (str) – Program file name (example: "myprogram.rxe").

Including this with mailbox I think wil be a usefull solution

schodet, 2022-01-02:
> I use RPI version 4. [...]

Thanks, I do not have a RPI4, but I will try to reproduce.

> for me everything seems to be OK [...]

Good :)

> I'm not sure what you meant with: here is a typo in your command, can you copy/paste?

You typed:

```
python3 nxt_test --log-level=debug --host=00:16:53:11:2D:2D
```

The address is wrong.

> When it comes to the two motor problem, my plan is as follow:
> create a sync motor program in labview, then copy the program to the NXT as an .rxe.
> According to the NXT-python API:
> Brick.start_program(name)
> Start a program on the brick.
> Parameters name (str) – Program file name (example: "myprogram.rxe").
> Including this with mailbox I think will be a usefull solution

Yes, this is an option. You can also use https://ni.srht.site/nxt-python/latest/api/motcont.html which does that with a program written in NXC.

Svein-Tore, 2022-01-02:
Argh, 2C of course:
```
pi@raspberrypi:~/.local/bin $ python3 nxt_test --log-level=debug --host=00:16:53:11:2D:2C
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/pi/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:16:53:11:2D:2C)
DEBUG:nxt.locator:found brick <nxt.brick.Brick object at 0xb641d9a0>
DEBUG:nxt.backend.bluetooth:send: 02:00:01:9b
DEBUG:nxt.backend.bluetooth:recv: 21:00
DEBUG:nxt.backend.bluetooth:recv: 02:9b:00:56:65:6c:6f:63:69:74:61:00:00:00:00:00:00:00:00:16:53:11:2d:2c:00:00:00:00:00:14:63:00:00
DEBUG:nxt.locator:found brick with name=Velocita and host=00:16:53:11:2D:2C
DEBUG:nxt.backend.bluetooth:send: 02:00:01:9b
DEBUG:nxt.backend.bluetooth:recv: 21:00
DEBUG:nxt.backend.bluetooth:recv: 02:9b:00:56:65:6c:6f:63:69:74:61:00:00:00:00:00:00:00:00:16:53:11:2d:2c:00:00:00:00:00:14:63:00:00
NXT brick name: Velocita
Host address: 00:16:53:11:2D:2C
Bluetooth signal strengths: (0, 0, 0, 0)
Free user flash: 25364
DEBUG:nxt.backend.bluetooth:send: 02:00:01:88
DEBUG:nxt.backend.bluetooth:recv: 07:00
DEBUG:nxt.backend.bluetooth:recv: 02:88:00:7c:01:1f:01
Protocol version 1.124
Firmware version 1.31
DEBUG:nxt.backend.bluetooth:send: 02:00:00:0b
DEBUG:nxt.backend.bluetooth:recv: 05:00
DEBUG:nxt.backend.bluetooth:recv: 02:0b:00:fe:1c
Battery level 7422 mV
Play test sound...DEBUG:nxt.backend.bluetooth:send: 06:00:80:03:2c:01:32:00
DEBUG:nxt.backend.bluetooth:send: 06:00:80:03:90:01:32:00
DEBUG:nxt.backend.bluetooth:send: 06:00:80:03:f4:01:32:00
DEBUG:nxt.backend.bluetooth:send: 06:00:80:03:58:02:32:00
done
INFO:nxt.backend.bluetooth:closing Bluetooth (00:16:53:11:2D:2C) connection
pi@raspberrypi:~/.local/bin $
```
Which means it worked this time.  Following up, I did
```
pi@raspberrypi:~/.local/bin $python3 nxt_test
Find brick...
NXT brick name: Velocita
Host address: 00:16:53:11:2D:2C
Bluetooth signal strengths: (0, 0, 0, 0)
Free user flash: 25364
Protocol version 1.124
Firmware version 1.31
Battery level 7450 mV
Play test sound...done
pi@raspberrypi:~/.local/bin $ 
```
So now it worked! Please don't ask me why!

I will look into Linus Atorf's script, though I don't feel it is straight forward.  

Thanks a lot for helping Nicolas!

schodet, 2022-01-03:
Thanks.

I will use this to improve the manual.

Closing.

closed 2022-01-03 by schodet